UbuntuAsahi/ubuntu-asahi #2: Is there any way to fix a broken bootup using your scripts?

rick-kilgore, 2022-12-15:
When I run it, I only see the ability to install.  My bootup is failing with the error messages shown in this pic of my screen during bootup: https://imgur.com/LVudPjm.

tobhe, 2022-12-15:
I believe you are hitting a bug with the current mesa version not working properly with an older kernel. You should be able to get a TTY with FN + CTRL + CMD + F2 and install linux-asahi-edge, which is a newer kernel that does work, from there.

tobhe, 2022-12-15:
I have pushed a new kernel so `apt update && apt upgrade` should also work

rick-kilgore, 2022-12-16:
Thank you, I will try that if I need to try with 22.10 again. But I am currently thinking that 22.10 is just too unstable for asahi.  I ran into a problem starting yesterday where Xorg crashes when I try to run it.  Is it still possible to run 22.04 instead?

tobhe, 2022-12-16:
I think the xorg issue on 22.10 is resolved. it was missing a 30-modeset.conf that is now shipped with the new asahi-configs package. I can reenable 22.04 if you prefer using that, but it I was not planning to backport the GPU driver.

rick-kilgore, 2022-12-16:
That's ok, I don't seem to need the GPU for what I'm doing.  I'll benefit it when the 22.10 changes are merged into the stable distro.

tobhe, 2022-12-16:
Alright, I have restored the old jammy images for now. I will upload new ones when I find the time to build and test them.

closed 2022-12-16 by tobhe