UbuntuAsahi/ubuntu-asahi #3: Add package: mesa-asahi

qwerzl, 2022-12-19:
Hi there,
Now that the experimental GPU driver is out, could you add mesa-asahi to the PPA?

tobhe, 2022-12-19:
Hi @qwerzl, there is a mesa package in the ppa and the gpu driver should work out of the box, see https://launchpad.net/~tobhe/+archive/ubuntu/asahi. All you need to do is upgrade to kinetic.

tobhe, 2022-12-20:
@sasa8810 If you install your ubuntu as documented in the [README](https://github.com/tobhe/ubuntu-asahi#readme) the PPA should be enabled by default. After the release upgrade you might be required to reenable it and change the version to the one you are running.

sasa8810, 2022-12-21:
So, it’s enabled by default?

qwerzl, 2022-12-21:
> Hi @qwerzl, there is a mesa package in the ppa and the gpu driver should work out of the box, see https://launchpad.net/~tobhe/+archive/ubuntu/asahi. All you need to do is upgrade to kinetic.

Ah I see. I was still in Jammy. Thanks for your work!

closed 2022-12-21 by qwerzl