UbuntuAsahi/ubuntu-asahi #4: While updating to Kinetic, the update magically failed to install and after a reboot, my Ubuntu install is f*cked up.

SamilKaraduman, 2022-12-24:
No grub boot menu, so not much I can do.

tobhe, 2022-12-24:
Any idea why it failed? where does it hang after reboot? In general a failed upgrade is pretty bad news, you might be able to switch to a tty via ctrl - alt - f2 and try to figure out what is wrong from there.

SamilKaraduman, 2022-12-24:
I'm really not sure, but here is a video clip.

SamilKaraduman, 2022-12-24:
https://user-images.githubusercontent.com/83664323/209448137-b23491f4-fea2-4eb0-8bb0-b2d3cf7a16ee.mov

tobhe, 2022-12-28:
This looks like some failure in the graphics pipeline, probably a bug in mesa in combination with the newest kernel. I am surprised because we had a similar bug at one point that I assumed was fixed with a recent update. As I said above, you should be able to switch to a tty and run `apt update && apt upgrade` by hand to fix it.

SamilKaraduman, 2022-12-29:
That did the trick. Thanks a lot!

closed 2022-12-29 by SamilKaraduman