UbuntuAsahi/ubuntu-asahi #5: Add macOS 13 boot firmware option to installer

TeaRex-coder, 2022-12-25:
After running the Asahi installer for 22.10 Ubuntu Desktop, the option for the boot firmware of macOS 13 is missing.

Screenshot attached
[image]

tobhe, 2022-12-28:
Thank you for the report! The installer is not very clear on this but I think the version here refers to the firmware revision which does not directly map to the macos version. Instead it seems to refer to the macos version that first introduced this firmware revision. The Asahi November status report at https://asahilinux.org/2022/11/november-2022-report/ suggests that it should be working fine if you select 12.3 with a M1 or 12.4 with a M2 machine.

TeaRex-coder, 2022-12-31:
It worked, thanks!

Do you know why the official installers at `curl https://alx.sh | sh` has 13?

closed 2022-12-31 by TeaRex-coder