UbuntuAsahi/ubuntu-asahi #6: Support encrypted home partition

tobiasgrosser, 2023-01-14:
Hi @tobhe, I just tried this installer and it worked very well. I had initially some trouble with spotty gpu driver on the M2, but the installation of asahi-edge and a reboot fixed this. As the system is working well, I am considering using it for my day-to-day work.

For this, I would like to encrypt my root partition. I wonder if it would be hard to add support for this to your ubuntu-asahi package? I am happy to redo my installation if that could help.

tobhe, 2023-01-15:
Thank you for the report! This is actually fairly high up on the todo list. What we need for this is a working USB installer image that installs the correct kernels and packages. The process will then be to use the `curl | bash` installer to get the boot loader chain up to u-boot and then use that to boot up the usb installer which supports all the features you get with the regular ubuntu install.

tobiasgrosser, 2023-01-15:
I looked a little further into this. Here are some relevant references. The asahi installer has a relevant issue https://github.com/AsahiLinux/asahi-installer/issues/137 but no further information.

There is some information about FDE on the first [asahi release post](https://asahilinux.org/2022/03/asahi-linux-alpha-release/) and their recommendation is to enable filevault and lock down u-boot and grub. That seems sensible, but I am unsure how to lock-down u-boot and grub and somehow would prefer to work towards disk encryption if I already start investigating.

> Apple Silicon machines are one of the few general purpose platforms that allows you to install your own custom OS while still maintaining a strong secure boot chain. Installing the bootloader requires physical access to the machine and your machine owner credentials (this is why we need to ask you to hold down the button to boot at the end of the install process!). Therefore, we are very interested in further supporting this in Linux in order to have a highly secure and attacker-resistant system, taking advantage of Apple’s SEP, Touch ID, and more, while still retaining full user control over their OS. We are designing the Asahi Linux boot process in order to allow this in the future, but the necessary bits aren’t ready yet, so please stay tuned!

> In the meantime, we recommend that users concerned about physical security enable FileVault in macOS. This will implicitly add a log-in requirement to recovery mode, which will prevent an attacker with physical access from being able to compromise your OS that way. m1n1 does not yet have a secureboot mode, but it also doesn’t have any local access features as long as it is installed properly. Locking down U-Boot and GRUB and the rest of Linux is left as an exercise for the user.

> The Asahi Linux installer does not have an option to set up FDE for you. However, you can use the UEFI-only option and roll your own traditional LUKS setup manually. We expect that users interested in advanced secure boot and encryption set-ups will do their own thing, at least for the time being.

There is also a recent [reddit post from this december](https://www.reddit.com/r/AsahiLinux/comments/zia9ec/what_is_the_status_of_full_disk_encryption_with/) where @marcan discusses details of his plans with respect to FDE on asahi. The options mentioned there are (a) a userspace implementation of luks in python to created the encrypted FS directly from OSX (b) install an encrypted-with-a-std-password volume and reencrypt in linux -- that seems fine but has the overhead of re-encryption. Marcan also discusses interesting [plans to avoid typing a pass phrase](https://www.reddit.com/r/AsahiLinux/comments/zia9ec/comment/j0l6mmi/?utm_source=share&utm_medium=web2x&context=3) by taking advantage of SEP and TouchID.

> The installer needs LUKS support. However, once SEP and TouchID are properly supported, we wouldn't want boring old LUKS passphrases anyway, so chances are the installer will set up some kind of dummy LUKS keyslot with a machine-tied key and no passphrase, and then you will be able to enable authentication with SEP/TouchID on Linux later (much like FileVault is technically always enabled, just using transparent machine keys) by replacing it with a wrapped key (this is similar to how Android full disk encryption worked before they replaced it with FBE).

There exists a fedora usb-drive image https://github.com/leifliddy/asahi-fedora-usb, which seems to use the usb-drive option that @tobhe also suggested (but without mentioning encryption).

I personally played with https://help.ubuntu.com/community/Installation/FromLinux#Debootstrap yesterday. The basic idea was to install a small ubuntu image of 10GB, then create a LUKS partition and debootstrap into it, then update grub to boot the LUKS partition. This works mostly fine. I can create the LUKS partition, install ubuntu, and open the partition in grub when booting. However, I am currently stuck with Linux not finding the root partition when actually booting. As I do some manual grub config editing, I probably miss some obvious things. With a little bit of debugging this should work, I guess.

There is also further information about the apple platform security at the end of this document: https://gist.github.com/chadmed/ea318ebb9574354cb44be5c31698dfc9

Among other things it states: 

> An existing weakness is that /boot must be stored in the clear, and there is currently no Secure Boot or Measured Boot analogue with which we can guarantee the integrity of the kernel or initramfs. 

I am a little confused. I had hoped we could have both /boot and / encrypted with luks and only the EFI partition would remain unencrypted. The ubuntu installer does not encrypt boot, but (https://help.ubuntu.com/community/Full_Disk_Encryption_Howto_2019) states this is possible.

I guess I currently am a little torn between (a) locking everything nicely down with SEP & TouchID or (b) getting a standard encrypted Ubuntu installation working. I guess a booting USB key would be nice as a start given that this is also the basis of the FDE howto mentioned above.

@tobhe, any idea what we would need to get this working?

tobiasgrosser, 2023-01-15:
Btw, here is a tool that for encrypting luks partitions from userspace: https://github.com/anisse/dmencrypt/blob/master/dmencrypt.go

tobiasgrosser, 2023-01-15:
Here the luks documentation of the on-disk format. https://mirrors.edge.kernel.org/pub/linux/utils/cryptsetup/LUKS_docs/on-disk-format.pdf.

I feel implementing his might not be too hard. In particular, one can test this easily on a Linux system.

tobiasgrosser, 2023-11-04:
@tobhe, I just finished a first prototype of a python-native implementation of luks that can turn an unencrypted filesystem image into an encrypted one. I wonder if that would allow the ubuntu/asahi installer to ask for a passphrase when installing the system:
https://github.com/opencompl/pyluks

Obviously, this does not yet give a fully authenticated bootchain, but this would give me the same level of security I have today on my laptop. If this is interesting, I would be glad to polish the tool further such that it can be integrated into the ubuntu asahi install process.

tobiasgrosser, 2023-11-17:
@tobhe, would you be able to give me a hint of how to get started to update ubuntu-asahi to support an encrypted boot partition? I wanted to try `sudo ./build-generic.sh` as indicated on the README, but it seems this does not exist anymore? If you have an idea how to approach this best, a rough dump of that idea would be much appreciated.

tobhe, 2023-11-21:
@tobiasgrosser indeed our build process has become a bit more complicated since we moved everything to launchpad. We need to update the README for that. The easiest way to build images locally seems to be using something like https://github.com/ubuntu-bartenders/ubuntu-old-fashioned. Images are built from livecd-rootfs which is available via git here https://github.com/UbuntuAsahi/livecd-rootfs/tree/ubuntu/mantic.

I wonder if you even need to build a new image though. I am hosting the installer images at https://files.tobhe.de/ubuntu/os/. It might be enough to unpack one of those, encrypt the root partition and pack it up again.

I don't know if there is an easy way to make a local file available via the installer but you can host it somewhere, download `https://ubuntuasahi.org/install`, point it to your own installer_data.json and REPO_BASE and run the installer.

robinkbr, 2024-05-20:
Hello, I was wondering if there were updates on encrypting the harddrive of asahi linux. Since one year ago @tobhe said it is quite high on the to do list I just wanted to know if a solution was found or if we still have to go through this risky guide: https://davidalger.com/posts/fedora-asahi-remix-on-apple-silicon-with-luks-encryption/

eindgebruiker, 2025-07-29:
For Fedora an apparently robust encryption script was created, based on David Alger's guide: https://github.com/osx-tools/asahi-encrypt
However there are some differences between Fedora and Ubuntu (use of dracut, btrfs instead of ext4, ...), so it cannot be used unchanged.

eindgebruiker, 2025-08-06:
FYI I was able to encrypt my Ubuntu Asahi 25.04 root partition using this guide: https://karthikkaranth.me/blog/setting-up-luks-encryption-on-an-existing-ubuntu-partition/ and the USB live image mentioned in https://github.com/UbuntuAsahi/ubuntu-asahi/issues/126