UbuntuAsahi/ubuntu-asahi #7: Chromium doesn't seem to work with wayland, can't find asahi driver

kqlski, 2023-01-22:
While using the 22.10 Desktop on my M1 Pro, I can't seem to manage to get wayland and chromium working (or rather chromium working at all).
Here is what i get with running `chromium --ozone-platform=wayland`:
[log.txt](https://github.com/tobhe/ubuntu-asahi/files/10474342/log.txt)

I tried quite a lot of things but couldn't manage to get it working.

Steps to reproduce:
1. `sudo snap install chromium`
2. `chromium --ozone-platform=wayland`

tobhe, 2023-01-23:
Thanks for the report, I have noticed this before but found no satisfying solution so far. My current hypothesis is that this happens because of the mesa version bundled with the snap which does not support the asahi gpu driver. Disabling the gpu sandbox via `--disable-gpu-sandbox` seems to get us around this crash but fails with a new error.

tobhe, 2023-01-23:
Another issue I encountered with snaps in particular is that they are being build with an older version of patchelf internally which produces non-16k-compatible binaries so the easiest way around this for now is probably getting a non-snap package.

kqlski, 2023-01-23:
Yeah, I figured that might be the issue but it was quite difficult find an arm64 version of chromium without building it myself.

kqlski, 2023-01-30:
Actually fedora seems to have a functional .rpm so this could be done with alien.

nullbert, 2023-02-08:
Hi! Did you try converting the .rpm? I just installed the 110 RC build, on the off chance it was built with an updated environment. No go...

nullbert, 2023-02-08:
Downloaded the .deb from debian with dependencies. Had some trouble with libevent, but forced the install and it works. I can now use the laptop as my daily driver. Thanks @tobhe!

vicjones, 2023-05-14:
Any updates here? I’ve been using —no-sanbox for my chromium based applications

tobhe, 2023-05-14:
I am currently waiting for the latest chromium beta at https://git.launchpad.net/~chromium-team/chromium-browser/+git/snap-from-source to hit arm64. With that we could get graphics support working out of the box by supplying our own patched gnome-sdk snap with asahi-mesa included.

I am still trying to figure out a distribution mechanism because snaps don't have a PPA equivalent and I don't know too much about snaps in general, but I have someone from the Ubuntu desktop team who might be willing to help us with that.

tobhe, 2023-06-18:
@clementperon that would be a solution but I am hesitant to do it because chromium comes with a huge maintenance burden. Maintaining our own would mean we have to seriously track security fixes and make sure we distribute them in time. Getting the regular version to work would give us all this for free.

clementperon, 2023-07-13:
```
sudo snap install chromium --beta
```
works without any hack :)

I have also enabled sync features following this tutorial: https://www.learningtopi.com/sbc/chromium-sync/

tobhe, 2023-07-13:
Yes, I can confirm that it does work with the newest beta version :) Thanks for sharing @clementperon!

eslerm, 2023-07-17:
After `sudo snap install chromium --beta` and launching chromium from command line I get the output:

```
MESA-LOADER: failed to open apple: /snap/chromium/2535/gnome-platform/usr/lib/aarch64-linux-gnu/dri/apple_dri.so: cannot open shared object file: No such file or directory (search paths /snap/chromium/2535/gnome-platform/usr/lib/aarch64-linux-gnu/dri, suffix _dri)
failed to load driver: apple
[67267:8:0713/193556.086415:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
```

`/snap/chromium/2535/gnome-platform/` does not exist `/snap/chromium/2535/usr/lib/aarch64-linux-gnu/dri/` does not contain `apple_dri.so`. Likely an issue in the upstream snap. Possibly related to https://github.com/UbuntuAsahi/ubuntu-asahi/issues/47

tobhe, 2023-07-17:
I think we need to build our own gnome-sdk snap with our mesa i+ driver ncluded. I remember talking to someone who managed to get that working for the Lenovo x13s which has a similar problem.

clementperon, 2023-08-05:
Ok so it fallback automatically on swrast on Chromium but not on Electron app right ?

clementperon, 2023-08-27:
With latest update snap gnome-42-2204 as change track to asahi/stable and it works fine!

Thanks a lot @tobhe

eslerm, 2023-10-29:
Thank you @clementperon  :)

closed 2023-10-29 by eslerm