UbuntuAsahi/ubuntu-asahi #8: When waken from sleep, GDM goes into 100% scaling, and after login, the keyboard, trackpad, and mouse stop working.

SamilKaraduman, 2023-01-28:
This mostly occurs when on battery power, but sometimes also occurs when the computer is plugged in. When I wake the computer from sleep, GDM is scaled to 100% and after I login, the keyboard trackpad, and mouse completely stop working. I then will have to perform a force reboot to fix the issue. But no matter what, I experience this issue at least once a day. Any suggestions? 
[dmesg.log](https://github.com/tobhe/ubuntu-asahi/files/10528339/dmesg.log)

https://user-images.githubusercontent.com/83664323/215292491-cbe3ca11-91b2-42e5-bf57-3bb08e01f174.mp4

tobhe, 2023-01-29:
Hi Samil, thanks for the report. I have seen this behavior before but it happens way less frequently than once a day on my m2. My best guess is that it is probably a kernel bug where some drivers fail to reattach on wakeup. I will move edge to a newer kernel version soon, let's hope that fixes it.

tobhe, 2023-01-31:
I pushed a few updates to the ppa. You could try to switch to the edge kernel via `apt install linux-asahi-edge` and see if that fixes your bug

SamilKaraduman, 2023-02-01:
I'll test the update right now. I'll close the issue later if everything goes well.

SamilKaraduman, 2023-02-01:
[image]

Unfortunately, after I've updated to the edge kernel Uboot now gives a strange error. The system also pretty much fails to boot up. I've restarted multiple times, but still nothing.

tobhe, 2023-02-01:
interesting, that should not happen. Does this persist after reboot? The only explanation i could think of would be that m1n1 or the device trees have not been updated properly.

SamilKaraduman, 2023-02-02:
Reboot after reboot after reboot, yet the issue still persists. Should I consider a reinstall?

tobhe, 2023-02-02:
Sorry for the inconvenience, here is how you might be able to restore your system by manually updating the bootloader from your macos partition:

I have uploaded the latest boot loader binary to my web server, you can fetch it with curl:
```
$ curl https://tobhe.de/ubuntu/boot.bin -o boot.bin
```
Make sure the checksum matches
```
$ shasum boot.bin
06d2e9409dbe6129ecc7ddbc552894e31cbd2026  boot.bin
```

Next find your Ubuntu EFI partition, this should be something like:
```
$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   7:                        EFI EFI - UBUNT             500.2 MB   disk0s7
```

mount it. Of course /dev/disk0s7 might be something different for you depending on the previous step.
```
$ mkdir mnt
$ sudo mount -tmsdos /dev/disk0s7 mnt
```
then copy the bootloader to disk and unmount
```
$ sudo cp boot.bin mnt/m1n1/boot.bin
$ diskutil unmount mnt
```
With that your Ubuntu SHOULD now boot again (i hope).

SamilKaraduman, 2023-02-02:
Kudos to you! My install of Ubuntu is booting once again! I'll be testing the edge kernel for a few hours now and see if it holds up. 

Once again, thank you very much!

tobhe, 2023-02-02:
Glad that it helped :) You should upgrade probably your packages to make sure this doesn't happen again! The cause was likely that your m1n1 was not up to date. I might have to add that as an explicit dependency for the kernel package to make sure they are always in sync.

SamilKaraduman, 2023-02-03:
Tested for a little while, and no weird things occurred during my testing. Thank you very much for your time and hard work!

closed 2023-02-03 by SamilKaraduman