UbuntuAsahi/ubuntu-asahi #9: Build did not output installer scripts

rayrayrayraydog, 2023-01-30:
I have run the build-generic.sh script on a Ubuntu 22.10 x64 machine and it has successfully created ubuntu.disk.img.zip, but there is no installer in the zip archive that I can find.  I noticed some umount errors from the script output near the end.  Can you help me find the installer?

```[install disk packages] Processing triggers for sgml-base (1.30) ...
[info] Synchronizing changes to disk
[info] Installing grub
[info] Generating grub image
[info] Adding user ubuntu
BAD PASSWORD: The password is shorter than 8 characters
[info] Cleaning up data...
umount: /mnt/data/code/ubuntu-asahi/build/rootfs.disk/var/cache/apt/archives: not mounted
+ ../scripts/disk/04-build-disk-image.sh
[info] Creating /mnt/data/code/ubuntu-asahi/build/ubuntu.disk.img
mke2fs 1.46.5 (30-Dec-2021)
Discarding device blocks: done                            
Creating filesystem with 1389568 4k blocks and 347440 inodes
Filesystem UUID: 87c6b0ce-3bb6-4dc2-9298-3a799bbb5994
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done 

[info] Mounting image
[info] Copying to disk
[info] Updating grub config
[info] Packing disk
[info] Compressing
  adding: disk.img (deflated 66%)
  adding: esp/ (stored 0%)
  adding: esp/m1n1/ (stored 0%)
  adding: esp/m1n1/boot.bin (deflated 72%)
  adding: esp/EFI/ (stored 0%)
  adding: esp/EFI/BOOT/ (stored 0%)
  adding: esp/EFI/BOOT/BOOTAA64.EFI (deflated 70%)
umount: /mnt/data/code/ubuntu-asahi/build/mnt: not found
builder@cat:/mnt/data/code/ubuntu-asahi $ ```

tobhe, 2023-01-30:
Right, you can get the script I am currently using via `curl https://tobhe.de/ubuntu/install`. For a local install the easiest way would probably be to spin up a local http server via something like python http.server and then pointing INSTALLER_DATA and REPO_BASE to your local server.

The directory layout for that would be sth like (where $www is your web server root):

```
$www/os/ubuntu-base.zip # Copy from build/ubuntu.disk.img.zip
$www/installer_data.json
```
Make sure the `package` field in installer_data.json matches the file name in `$www/os`

I will see if I can add more documentation and the actual install script I'm using to the repo.

rayrayrayraydog, 2023-01-30:
Thank you for your response.  I will take a look into that.  Unfortunately I had gotten ahead of myself and didn't realize Asahi doesn't support the new M2 mini yet.  :/

tobhe, 2023-02-06:
That is sad to hear but I am optimistic that it will work soon! It looks like a few asahi developers have already gotten their hands on dev machines.
I am closing this bug for now since I added [install.sh](https://github.com/tobhe/ubuntu-asahi/blob/kinetic/install.sh) to the repo.

closed 2023-02-06 by tobhe