UbuntuAsahi/ubuntu-asahi #10: apt upgrade makes ubuntu 22.04 unbootable

kindlychung, 2023-02-13:
I installed ubuntu 22.04 and everything worked until I tried `sudo apt update && sudo apt upgrade`, which results in some failure in grub efi configuration. After a restart I could not boot into ubuntu any more. The error looks like this:

```
Scanning nvme 0:4...
Unable to read file /
Failed to load /
BootOrder not defined
EFI boot manager: cannot load any image
Found EFI removable media binary efi/boot/...
180224 bytes read in 0 ms
Booting /efi\boot\bootaa64.efi
error: no such device: 87c6b...
error: unknown filesystem.
Entering rescue mode...
grub rescue>
```

MacOS is still working though.

tobhe, 2023-02-13:
was this after the first reboot? there used to be a race in the first-boot script that broke the random uuid renaming, but that should be fixed. This error message looks suspicious because the default uuid that gets changed with the first-boot script is ROOT_UUID=87c6b0ce-3bb6-4dc2-9298-3a799bbb5994 which seems to match your grub config.
I wonder if this happens again if you try a second time. The previous bug used to happen randomly.

kindlychung, 2023-02-14:
Yeah, it's the first reboot. 

Unfortunately I have wiped it out so won't be able to try a second time. But I might give 22.10 a go. Thanks for your work!

nreef12, 2023-03-26:
I'm having about the same issue on the first reboot, it does seem like it has to do with the UUID [image]

tobhe, 2023-03-26:
Yes, there is a first-boot script that is supposed to change the uuid from the hardcoded initial one to a random one. It looks like it sometimes fails which causes a mismatch between what grub expects the actual UUID. This seems to happen randomly so I don't really know how to reproduce and fix it reliably. I thought I had fixed it at some point but apparently it just never happened on my hardware anymore.

tobhe, 2023-03-26:
This [reddit post](https://old.reddit.com/r/AsahiLinux/comments/121eg0j/stuck_on_this_screen_ho_a_do_i_fix/) suggests that it actually might not be a Ubuntu specific problem but also happen on regular Asahi.

nreef12, 2023-03-28:
> This [reddit post](https://old.reddit.com/r/AsahiLinux/comments/121eg0j/stuck_on_this_screen_ho_a_do_i_fix/) suggests that it actually might not be a Ubuntu specific problem but also happen on regular Asahi.

I tried @marcan's suggestion on that post to run `sudo update-grub`, but it just tells me this:

```
grub probe: error: failed to get canonical path of `/dev/disk/by-uuid/87c6b0ce-3bb6-4dc2-9298-3a799bbb5994
```

nreef12, 2023-03-28:
Also, running `journalctl -b0 > log.txt` gets [this](https://pastebin.com/FGtAzBGy)

eslerm, 2024-02-14:
Ubuntu Asahi has moved to 9 months of Ubuntu release support. Please ask to re-open issue if a supported version is affected.

closed 2024-02-14 by eslerm