UbuntuAsahi/ubuntu-asahi #11: Unable to Install – no OS selection

Rastafabisch, 2023-02-15:
I cannot install anything as I'm not offered any choice. I typed in `Ubuntu 22.10 Desktop` and similar but this led to no avail.
[image]

tobhe, 2023-02-15:
Thanks for the report, I have not seen this before. What kind of hardware are you using? Is this one of the new M2 machines?

Rastafabisch, 2023-02-15:
I am using a MacBook Pro M1 Pro. I have done some testing (comparing it to stalk Asahi and think now, that the issue is, that I try to install on an external drive. I suppose there is now support as of now, and it will take some time until "upstream" Asahi implements this, so feel free to close the issue. Sorry for the noise.

tobhe, 2023-02-15:
That would make sense, I have never tried that. It might be worth to make the installer show a clearer error message in this case, but that would need to be done in https://github.com/AsahiLinux/asahi-installer

closed 2023-02-15 by tobhe