UbuntuAsahi/ubuntu-asahi #12: Unbootable system after kernel upgrade

shortontech, 2023-02-16:
I've installed (and wiped) from this repo 3 or four times, mostly getting the error from #10 

If I select the the older kernel the system loads up but the trackpad and keyboard both don't function, which at least means that #10 doesn't seem to be as much of an issue (for me) anymore.

Thank you for maintaining the product, let me know if you need any information or testing later on. Here is what's displayed on the screen as accurately as I could type it.

```
[    0.287011] apple-pmgr-pwrstate 28e0000.power-management:power-controller02e8: failed to add parent domain: -22 (power-controller -> power-controller)
[    0.287071] apple-pmgr-pwrstate 28e0000.power-management:power-controller02e8: failed to add parent domain: -22 (power-controller -> power-controller)
[    0.327132] apple-dcp 38bc00000.dcp dcp_poweron() starting
Gave up waiting for root file system device. Common problems:
 - Boot args (car /proc/cmdline)
 - Missing modules (cat /proc/modules; ls /dev)
 - ALERT! UUID=5aaa4a0b-447c-497c-8734-76f31bbfe57b does not exist. Dropping to a shell!```

tobhe, 2023-02-16:
Which version of are you installing and what hardware are you using? UUID=5aaa4a0b-447c-497c-8734-76f31bbfe57b again sounds like the first-boot script might be the problem. I was planning to release new images soon, maybe the newest build works better.

shortontech, 2023-02-16:
I'm not sure which number I need to get you. I'm looking at the `dpkg -l`
The kernel that doesn't boot as `linux-asahi` is kernel `6.2.0.8`
`ubuntu-asahi` is at `1.9`
`asahi-configs` is `1-2`
`asahi-scripts` is `20220821-16`
`u-boot-asahi` is version `2022-10-1`

Since both the older kernels are both installed, `linux-image-6.1.0.6-asahi` is the one that still boots.

Since one kernel works and the other doesn't, I figure it may be a kernel issue, but this is one that doesn't present itself when running the original arch based version of asahi.

If you want I can wipe and reinstall the arch version, make sure the kernel is updated and see if the issue persists on there.

tobhe, 2023-02-16:
From what I see I am assuming this is 22.10. This might be a kernel or a device tree problem. There are two things you could try: 1. run `sudo updadte-m1n1` to update your bootloader and see if that changes anything.
2. try installing an older version like 6.1.0-7.3 from https://launchpad.net/~tobhe/+archive/ubuntu/asahi/+sourcepub/14449233/+listing-archive-extra

Other versions I would be interested in are m1n1 and mesa

shortontech, 2023-02-16:
You are correct in assuming this is 22.10

The package for `m1n1` is currently `1.2.4-1` and most of the mesa packages are `23.0.0-1asahi2~pre20221214` except for `libglu1-mesa` which is `9.0.2-1`.

I'm using the a 14 inch macbook pro, though I'm not clear on the processor, which probably only matters if it's a kernel issue.

1. I ran `sudo update-m1n1` and tried all three kernel versions several times without success.
2. I tried 6.1.0-7.3 and it hung on loading the ram disk.

Here's the current list of installed packages from `dpkg -l | grep asahi`
```
ii  alsa-ucm-conf-asahi                        1-1                                      arm64        ALSA Use Case Manager configuration (and topologies) for Apple silicon devices
ii  asahi-configs                              1-2                                      arm64        Asahi Linux misc configs
ii  asahi-scripts                              20220821-16                              arm64        Asahi Linux maintenance scripts
ii  libegl-mesa0:arm64                         23.0.0-1asahi2~pre20221214               arm64        free implementation of the EGL API -- Mesa vendor library
ii  libgbm1:arm64                              23.0.0-1asahi2~pre20221214               arm64        generic buffer management API -- runtime
ii  libgl1-mesa-dri:arm64                      23.0.0-1asahi2~pre20221214               arm64        free implementation of the OpenGL API -- DRI modules
ii  libglapi-mesa:arm64                        23.0.0-1asahi2~pre20221214               arm64        free implementation of the GL API -- shared library
ii  libglx-mesa0:arm64                         23.0.0-1asahi2~pre20221214               arm64        free implementation of the OpenGL API -- GLX vendor library
ii  libunwind8:arm64                           1.6.2-0ubuntu1+asahi.0                   arm64        library to determine the call-chain of a program - runtime
ii  linux-asahi                                6.2.0.8.8                                arm64        Complete Asahi Linux kernel and headers
ii  linux-asahi-headers-6.1.0-6                6.1.0-6.2                                arm64        Header files related to Linux kernel version 6.1.0
ii  linux-asahi-headers-6.1.0-7                6.1.0-7.3                                arm64        Header files related to Linux kernel version 6.1.0
ii  linux-asahi-headers-6.2.0-8                6.2.0-8.8                                arm64        Header files related to Linux kernel version 6.2.0
ii  linux-asahi-tools-6.1.0-7                  6.1.0-7.3                                arm64        Linux kernel version specific tools for version 6.1.0-7
ii  linux-buildinfo-6.1.0-7-asahi              6.1.0-7.3                                arm64        Linux kernel buildinfo for version 6.1.0 on ARMv8 SMP
ii  linux-headers-6.1.0-6-asahi                6.1.0-6.2                                arm64        Linux kernel headers for version 6.1.0 on ARMv8 SMP
ii  linux-headers-6.1.0-7-asahi                6.1.0-7.3                                arm64        Linux kernel headers for version 6.1.0 on ARMv8 SMP
ii  linux-headers-6.2.0-8-asahi                6.2.0-8.8                                arm64        Linux kernel headers for version 6.2.0 on ARMv8 SMP
ii  linux-headers-asahi                        6.2.0.8.8                                arm64        Asahi Linux kernel headers
ii  linux-image-6.1.0-6-asahi                  6.1.0-6.2                                arm64        Linux kernel image for version 6.1.0 on ARMv8 SMP
ii  linux-image-6.1.0-7-asahi                  6.1.0-7.3                                arm64        Linux kernel image for version 6.1.0 on ARMv8 SMP
ii  linux-image-6.2.0-8-asahi                  6.2.0-8.8                                arm64        Linux kernel image for version 6.2.0 on ARMv8 SMP
ii  linux-image-asahi                          6.2.0.8.8                                arm64        Asahi Linux kernel image
ii  linux-modules-6.1.0-6-asahi                6.1.0-6.2                                arm64        Linux kernel extra modules for version 6.1.0 on ARMv8 SMP
ii  linux-modules-6.1.0-7-asahi                6.1.0-7.3                                arm64        Linux kernel extra modules for version 6.1.0 on ARMv8 SMP
ii  linux-modules-6.2.0-8-asahi                6.2.0-8.8                                arm64        Linux kernel extra modules for version 6.2.0 on ARMv8 SMP
ii  linux-modules-extra-6.1.0-6-asahi          6.1.0-6.2                                arm64        Linux kernel extra modules for version 6.1.0 on ARMv8 SMP
ii  linux-modules-extra-6.1.0-7-asahi          6.1.0-7.3                                arm64        Linux kernel extra modules for version 6.1.0 on ARMv8 SMP
ii  linux-modules-extra-6.2.0-8-asahi          6.2.0-8.8                                arm64        Linux kernel extra modules for version 6.2.0 on ARMv8 SMP
ii  linux-tools-6.1.0-7-asahi                  6.1.0-7.3                                arm64        Linux kernel version specific tools for version 6.1.0-7
ii  mesa-vulkan-drivers:arm64                  23.0.0-1asahi2~pre20221214               arm64        Mesa Vulkan graphics drivers
ii  u-boot-asahi                               2022.10-1                                arm64        A boot loader for Apple silicon systems
ii  ubuntu-asahi                               1.9                                      arm64        Packages for Apple Silicon
```

tobhe, 2023-02-17:
I don't see anything obviously wrong here.
> I ran sudo update-m1n1 and tried all three kernel versions several times without success.

So now 6.1.0.6-asahi which worked previously is also broken?

It looks like there either is a kernel bug triggered by your specific hardware (m2 air and m1 mini still work) or some weirdness going on with your device trees. What does `ls /lib/firmware/*/device-tree` show for you?

a-plastic-bag, 2023-02-17:
Same issue here (M1 Pro).
Kernel `6.2.0-8-asahi` fails to boot with same error.
kernel `6.1.0-6-asahi` boots successfully but, as said above, keyboard and trackpad are hit or miss.
Tried `sudo update-m1n1` to no avail.

tobhe, 2023-02-17:
I am fairly sure at this point that it is a kernel bug triggered by the m1 pro. What you could do for now is boot into the old kernel, remove the broken kernel with `sudo apt remove linux-image-6.2.0-8-asahi` and then `apt-mark hold linux-image-asahi` to prevent it from upgrading to the broken version.

I will try to build a new kernel based on [v6.2-rc8](https://github.com/AsahiLinux/linux/releases/tag/v6.2-rc8) which was tagged 4 days ago and let you know once you can test that to see if it is fixed there.

tobhe, 2023-02-17:
fwiw I have unlisted the kernel update for now and restored 6.1 so that hopefully no one else will run into this. Thank you all for the feedback.

Rastafabisch, 2023-02-26:
Apparently the same bug is still valid for `linux-image-6.2.0-8-asahi-edge`, thus in this case currently also disabling the ability to receive graphics acceleration.

tobhe, 2023-02-26:
Hi @Rastafabisch, thank you for the report. Indeed this bug has not been fixed in the `-edge` kernel. However, both `asahi` and `asahi-edge` have graphics acceleration enabled, the only difference is the upstream kernel version they are based on. This is different than in the asahi distribution where non-edge still has graphics disabled. The reason for this is that we had a bug where updated mesa and kernel without graphics driver would not work, so it was fixed by enabling the graphics driver in all kernels.

Rastafabisch, 2023-02-26:
Then this is certainly off-topic and I can open a new bug report for this, but I do not habe graphics acceleration. The information panel in the system settings states that only software acceleration (using X11) is available.

tobhe, 2023-02-26:
Sure, new bug report sounds like a good idea.

tobhe, 2023-02-28:
@a-tringali can you paste the output of `uname -a` and `apt list --installed | grep asahi` from your machine. You might need to downgrade from `asahi-edge` to regular `asahi`.

miyuanbo, 2023-03-10:
> fwiw I have unlisted the kernel update for now and restored 6.1 so that hopefully no one else will run into this. Thank you all for the feedback.
> 
> EDIT: of course that means you won't need to mark `linux-image-asahi` for hold anymore. Booting an older kernel and then uninstalling 6.2.0-8.8 should be enough to fix it

Well, I applied "sudo apt upgrade" yesterday, and the kernel version was updated to 6.1.0-1. 
However, a huge lag in touchpad occurs, same as "kernel 6.1.0-6-asahi boots successfully but, as said above, keyboard and trackpad are hit or miss" as is said above.

miyuanbo, 2023-03-13:
@tobhe Oh, I'm afraid I was on 22.04. I'll switch to 22.10 then, thanks!!!

miyuanbo, 2023-03-13:
I've tried two times of clean install of 22.10 (Deleted all of the partitions, then re-run the install.sh script), but both just failed to boot up, only showing blank screen (all black) after the EFI prints the line "EFI: booting boot.efi". The green LED on Caps-lock will light-up on press, though.

miyuanbo, 2023-03-14:
I tried 22.10 both Desktop and Server versions, clean install, but u-boot both prints like this.
[image]

tobhe, 2023-03-16:
So 22.04 works but 22.10 does not? This is new to me... I will try to figure out what might have changed there and why you are getting a different result than others. Is your macos partition up to date? This could be a firmware issue. Anyway it might be better to move this discussion to a separate issue.

miyuanbo, 2023-03-17:
Okey, I'll try a few more installations.
I use the "diskutil" command in macos to delete all the 3 partitions created by the asahi-installer (2.5G-apfs & 500M-EFI & 50G-LinuxRoot) after each failed install, does this make my "macos partition up to date"?
The successful installation of 22.04 was the first try for Ubuntu, but before this I had stock asahi linux (Arch Linux) installed.
Then I wiped the 22.04 with "diskutil" and tried 22.10, failed for 3 times, including 22.10 Desktop and Server.
I will try 22.04 again, if it successes, maybe the problem is within the 22.10; otherwise it may be my own problem.

op06072, 2023-03-17:
I have same problem. But I can boot on normal asahi linux kernel. However I can't boot on asahi-edge kernel.

op06072, 2023-03-17:
> I have same problem. But I can boot on normal asahi linux kernel. However I can't boot on asahi-edge kernel.

Ah maybe, it's cause that asahi-edge kernel is 6.2.0-8 and normal kernel is 6.2.0-7.

tobhe, 2023-03-24:
I have uploaded a few updates including a new asahi-edge kernel for kinetic. I'm hoping that this one fixes the bugs we saw previously but as I don't have the hardware to trigger it, it would be nice if someone else could test it. Make sure to update all your packages before installing linux-asahi-edge.

miyuanbo, 2023-04-04:
Tried the newest version of 22.10 desktop, it boots!
Thanks for the work!

closed 2024-01-14 by tobhe