UbuntuAsahi/ubuntu-asahi #13: Failed boot due to disk not present

JumpMaster, 2023-02-21:
Hi,

I previously has Asahi linux installed and removed all partitions and container volumes before starting.  No issues during the install as per the previous Asahi install.  I even tried this at least 3 times and got the same result.  Photo below.

macOs 13.2.1
Ubuntu 22.10

[image]

Thanks!

tobhe, 2023-02-21:
Looks like this is the same bug as https://github.com/tobhe/ubuntu-asahi/issues/12. Is this a m1 macbook pro by any chance? I think I know how to fix it (I probably need to regenerate the disk image).

JumpMaster, 2023-02-21:
Yes your spot on! Would be great to give this a try. Thanks

tobhe, 2023-02-21:
New images are already building. I'll let you know once they are uploaded.

tobhe, 2023-02-21:
22.10 desktop should now work again!

JumpMaster, 2023-02-21:
Yep it's now working!  Had a weird black screen on first boot. After a second boot I'm in.  Thanks!

tobhe, 2023-02-22:
Awesome, thanks for the feedback, glad that it worked! Not sure what caused the black screen but first-boot can be a bit flaky sometimes.

closed 2023-02-22 by tobhe

GjergjiFloychi, 2023-02-22:
Lucky me installed this right now. Is there any way to fix it faster than the time it takes to reinstall it from scratch? I'm on the mac m1 air base model.

tobhe, 2023-02-22:
> Lucky me installed ubuntu 22.04 half an hour ago. Is there any way to fix it faster than the time it takes to reinstall it from scratch? I'm on the mac m1 air base model.

The only thing that would work is booting an older kernel, but since this is a new install you probably don't have one on your disk.
I think reinstalling is the safest and fastest method in your case.

GjergjiFloychi, 2023-02-22:
I will reinstall the Ubuntu 22.04 version and tell you if it works. Alternatively since it only works once on the first boot installing an older kernel is a possibility.

GjergjiFloychi, 2023-02-22:
It works fine on the old version, as expected the edge version does not boot after the first time.