UbuntuAsahi/ubuntu-asahi #14: Only software acceleration available

Rastafabisch, 2023-02-26:
I'm running Ubuntu 22.10 (upgraded from 22.04) on my MacBook Pro M1 Pro. I do not have any graphics acceleration available.
The information panel within system settings reflects this, taking that the OS uses software-acceleration on the X11 winding system. 22.04 previously also stated that it's using software acceleration using llvmpipe, thought it had been running on Wayland.
I do have display brightness control available though (on Asahi those are connected as both are only available on the edge kernels.)

tobhe, 2023-02-26:
Can you share the contents of your `/etc/apt/sources.list.d/asahi-ppa.list`?
It looks like you might still be using an old 22.04 kernel and mesa version.

tobhe, 2023-02-27:
Other interesting outputs would be `apt list --installed | grep asahi` and `uname -a`.

Rastafabisch, 2023-02-27:
I ran those commands and I am attaching the output. 
```
cat /etc/apt/sources. list.d/asahi-ppa.list
# Ubuntu packages for Apple Silicon hardware
de [signed-by=/usr/share/keyrings/tobhe-asahi.gpg]https://ppa.launchpadcontent.net/tobhe/asaht/ubuntu jammy main
```
This one is properly already the issue. How would I update this to reflect 22.10 "Kinetic" instead of 22.04 "Jammy"

```
apt list -installed | grep asahi
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
alsa-ucn-conf-asahi//janny.kinetic,now1-1arm64[installed,autonatic]
asahí-configs/janny,kinetic,now1-2arn64[installed,autonatic]
asahi-scripts/janny,kinetic,now20220821-16arn64[installed,autonatic].
libegl-mesa/kinetic,now23.0.0-1asahi2-pre20221214arn64[installed,automatic]
libgbr1/kinetic,now23.0.0-1asahi2-pre20221214arm64[installed,autonatic]
libgl1-nesa-dri/kinetic,now23.0.0-1asahi2-pre20221214arn64(installed,automatic]
libglapi-mesa/kinetic,now23.0.0-1asahi2-pre20221214arm64[installed,automatic]
libglx-mesa0/kinetic,now23.0.0-1asahi2~pre20221214arn64[installed,automatic]
libunwind8/kinetic,now1.6.2-Oubuntu1tasahi.Oarm64.[installed,autonatic]
linux-asahi-edge-headers-6.1.1-1/jammy-kinetic,now6.1.1-1.0arm64[installed,automatic]
Linux-asahi-edge-headers-6.2.0-8/kinetic,now6.2.0-8.8arn64[installed,auto-renovable]
linux-asahi-headers-6.1.0-7/kinetic,now 6.1.0-7.3 arn64 [installed, autonatic] linux-asaht/kinetic, now 6.1.0.7.3 arm64 [installed, automatic] linux-headers-6.1.0-7-asaht/kinetic,now6.1.0-7.3arn64[installed,automatic]
Linux-headers-6.1.1•1-asahi-edge/janny,kinetic,now6.1.1-1.0arm64[installed,automatic]
linux-headers-6.2.0-8-asahi-edge/kinetic,now6.2.0-8.8arm64[installed,auto-removable]
linux-headers-asaht-edge/kinetic,now6.2.0.8.8arm64[installed,auto-renovable]
Linux-headers-asahi/kinetic,now6.1.0.7.3arn64[installed,automatic]
Linux-image-6.1.0-7-asaht/kinetic,now 6.1.0-7.3 arm64 [installed, automatic] linux-image-6.1.1-1-asaht-edge/jammy,kinetic,now6.1.1-1.0arn64[installed,autonatic] linux-inage-asaht/kinetic,now6.1.0.7.3arm64[installed,automatic]
linux-modules-6.1.0-7-asaht/kinetic,now 6.1.0-7.3 arm64 (installed, autonatic]
Linux-modules-6.1.1-1-asaht-edge/janny,kinetic,now6.1.1-1.0arm64[installed,automatic]
Linux-nodules-extra-6.1.0-7-aseht/kinetic,now6.1.0-7.3arm64[installed,autonatic]
linux-modules-extra-6.1.1-1-asahi-edge/jammy,kinetic,now6.1.1-1.0arm64(installed,autonatic]
nesa-va-drivers/kinetic,now23.0-0-lasaht2-pre20221214arm64[installed,autonatic]
nesa-vdpau-drivers/kinetic,now23.0.0-lasahi2-pre20221214arn64(installed,automatic).
nesa-vulkan-drivers/ktnetic,now23.0.0-1asaht2~pre20221214arn64[installed,automatic]
u-boot-asaht/jammy,kinetic,now2022.10-1arm64[installed,automatic]
ubuntu-asahi/jammy,kinetic,now 1.9 arm64 [installed]
```

```
uname -a
Linux ubuntu 6.1.1-1-asaht-edge #0-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 29 12:38:39 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

tobhe, 2023-02-27:
All you need to do is update the line to:
```
deb [signed-by=/usr/share/keyrings/tobhe-asahi.gpg]https://ppa.launchpadcontent.net/tobhe/asaht/ubuntu kinetic main
``` 
The ubuntu release upgrade process seems to have problems dealing with PPAs unfortunately.

Rastafabisch, 2023-02-27:
Apparently this does not seem to be enough. Maybe some `dpkg` command? Or shall I start new directly installing 22.10?

tobhe, 2023-02-27:
Reinstalling should not be necessary. What I don't understand is why uname reports 6.1.1-1-asahi-edge without linux-image-6.1.1-1-asahi-edge being in the list of installed packages. Maybe try removing it explicitly with `sudo apt remove linux-image-6.1.1-1-asahi-edge` then run `sudo update-m1n1` and reboot.

tobhe, 2023-02-27:
Awesome, glad I could help!

closed 2023-02-27 by tobhe