UbuntuAsahi/ubuntu-asahi #15: New Mesa version

adam44gg, 2023-03-22:
Hi, I was wondering when https://launchpad.net/~tobhe/+archive/ubuntu/asahi will have the updated Mesa version just released recently https://asahilinux.org/2023/03/road-to-vulkan/. 

I am also curious on how I could build the Asahi-edge kernel and Mesa myself if possible.

tobhe, 2023-03-22:
Hi, I am still working on the new kernel and mesa version. Keep in mind that you will need to update both and possibly m1n1 to get a working system.

If you really want to try it yourself (at your own risk) you can try looking at the asahi PKGBUILD sources at https://github.com/AsahiLinux/PKGBUILDs/ which should help you find the source code for each of those.
Waiting a few days for me to package it all might be easier though.

tobhe, 2023-03-24:
fwiw I have a first version working, see https://launchpad.net/~tobhe/+archive/ubuntu/asahi-testing
I will probably release them in the next days.

tobhe, 2023-03-25:
Now available. For now you have to install `linux-asahi-edge` to test it.

closed 2023-03-25 by tobhe