UbuntuAsahi/ubuntu-asahi #24: Add Ubuntu Flavors

tobhe, 2023-05-08:
With more of the build process moving to launchpad https://github.com/tobhe/ubuntu-asahi/issues/23 building Ubuntu flavor installs seems relatively easy. The main problem I see is that more disk images means more testing but I think it would be cool to have a kubuntu or cinnamon variant available out of the box.

tobhe, 2023-06-19:
Update: We can now build flavors with our regular image build process. I have successfully tested Xubuntu, Kubuntu and Ubuntu Cinnamon. The only one I didn't get to build was Ubuntu Studio because it pulls in snaps that aren't packaged for arm64.

We should probably do some user research to figure out if there is any real demand and possibly also talk to the flavor upstream devs before we commit to shipping images.

eslerm, 2023-06-20:
I'd like to support Kubuntu heavily.

Could we add a `Flavors` or `Other Flavors` folder during the install to reduce clutter?

iiuc, Studio configures low latency JACK and runs a modified [low latency kernel](https://help.ubuntu.com/community/UbuntuStudio/RealTimeKernel). Perhaps only a limited set of packages could be installed? I would like it for low latency configurations, MIDI, and a snap install of [Orca](https://hundredrabbits.itch.io/orca).

tobhe, 2023-06-20:
> I'd like to support Kubuntu heavily.
>
> Could we add a Flavors or Other Flavors folder during the install to reduce clutter?

We would have to fork the asahi installer and change a few things there as it doesn't support any kind of tree structure at the moment, only a list of entries.
I think Kubuntu would be really cool but we should make sure not to accidentally DOS the project with user questions. 

> iiuc, Studio configures low latency JACK and runs a modified [low latency kernel](https://help.ubuntu.com/community/UbuntuStudio/RealTimeKernel). Perhaps only a limited set of packages could be [...]

The low latency kernel in studio is another problem but one we can fix, not sure what we can do about the snaps. Ideally I think we should open bugs and ask them to release arm64 versions.

eslerm, 2023-06-20:
Perhaps instead of a folder, it could be a menu for distro/flavor and a menu for version.

alexellis, 2024-07-16:
Hi @tobhe I use two Mac Mini M1s with Arch Linux, however this is now discontinued, and I've found Go programs do not work properly on Fedora remix, segfaulting.

I only need headless access and don't want any kind of X11 or UI packages. Would you consider adding an image for Ubuntu server 24.04?

tobhe, 2024-07-16:
@alexellis certainly yes, I have that on my todo list. It will need a little more work since I want to have proper Ubuntu server builds as produced by launchpad instead of rolling our own. This means we have to patch the upstream build tooling in a few places as we did for UbuntuAsahi desktop.

alexellis, 2024-07-16:
Thanks for the response.

In lieu of the Ubuntu server build, is there a script or approach for "de-desktoping" the existing installation? Not ideal, but may provide a stop-gap.

tobhe, 2024-07-16:
I don't think there is a script but most stuff should be pulled in by the `ubuntu-desktop` and `ubuntu-desktop-minimal` meta packages. You can probably install `ubuntu-server`, remove those two and then run apt autoremove to get something close to regular server builds.