UbuntuAsahi/ubuntu-asahi #26: Build a working Live Install USB image

tobhe, 2023-05-08:
Having a working live installer would allow us to utilize all the install time configuration options that gives us such as custom partitioning, root on zfs and full disk encryption.

We would still have to provide a `curl | bash` install script to run in macOS to set up everything up to u-boot because Apple does not natively support USB booting (and we need access to the firmware shipped in macOS).

tobhe, 2023-05-08:
An initial live image is available at https://files.tobhe.de/. It does not properly handle firmware loading and the installer does not actually work but it is good enough to use as last resort to fix a broken system or to manually set up full disk encryption.

rakshith-ravi, 2023-06-13:
> An initial live image is available at https://files.tobhe.de/. It does not properly handle firmware loading and the installer does not actually work but it is good enough to use as last resort to fix a broken system or to manually set up full disk encryption.

Hi @tobhe. Thank you so much for a live image. However, I'm not entirely sure how to use this image though :sweat_smile:. Could you please help me with the instructions on how to use this?

tobhe, 2023-06-13:
@rakshith-ravi That probably depends on what you are trying to achieve.

The image needs to be written to a usb drive with dd like any regular USB installer.  To use the USB image you will need to have the regular Ubuntu Asahi or any kind of Asahi Linux installation up to the u-boot bootlader installed on your system. You can then boot from USB with the `run bootcmd_usb0` command in the u-boot prompt.

As explained above this is not the regular way to use Ubuntu Asahi, the image is really only useful to have a system rescue environment.

tobhe, 2023-06-13:
Ah makes sense. There have been successful attempts to do that with the live disk but it isn't trivial. Basically boot from live disk, set up your encrypted disks and then manually set up the root disk in the cryptlvm. You will likely also need to make /boot a separate partition.