UbuntuAsahi/ubuntu-asahi #28: Document uninstall process in README/website

tobhe, 2023-05-08:
We currently don't have any documentation on how to remove Ubuntu-Asahi or how to repair a failed installation. There should be a detailed easy to follow guide on how to do that, probably including links to the relevant upstream Asahi docs.

nreef12, 2023-05-08:
> probably including links to the relevant upstream Asahi docs
Definitely include this guide: https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet

bbbbbb11, 2023-05-09:
Usefull screenshot.
File system after / before installation

[image]

cowpod, 2023-05-12:
What worked for me was booting into recovery, and running the wipe-linux.sh script provided by the Asahi team.
Running it in normal OS resulted in permission denied errors, and leftover partitions (such as the EFI partition).

tobhe, 2023-05-12:
@cowpod Thanks, I almost forgot about that. Maybe I should mirror that on my domain too. I vaguely remember the Asahi docs warning against using it blindly and I don't know how well it plays with multiple Linux installations.

clementperon, 2023-06-18:
@tobhe at the moment the script will erase a random APFS container where the size match 2.5GB, and volumes that match 'asahi|linux|EFI'.
https://github.com/AsahiLinux/asahi-installer/blob/main/tools/wipe-linux.sh

I would have been interested to know that it exist before manually wiping my previous ArchLinux manually :)

It could be improved a bit by adding a user confirmation before performing these operations.

cowpod, 2023-07-17:
The script warned you what it'd do when you ran the script. 

```
echo "THIS SCRIPT IS DANGEROUS!"
echo "DO NOT BLINDLY RUN IT IF SOMEONE JUST SENT YOU HERE."
echo "IT WILL INDISCRIMINATELY WIPE A BUNCH OF PARTITIONS"
echo "THAT MAY OR MAY NOT BE THE ONES YOU WANT TO WIPE."
echo
echo "You are much better off reading and understanding this guide:"
echo "https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet"
echo
echo "Press enter twice if you really want to continue."
echo "Press Control-C to exit."
```

We could get the partition UUIDs using like `diskutil info /dev/diskXsX | grep "Disk / Partition UUID" | sed 's/.*:     //'` and place them somewhere, eg. in the EFI partition. Then when we run the uninstall script it'll delete partitions by UUID.
Downside: If we somehow delete the EFI partition, we'd lose those UUIDs. This is unlikely, and if we do lose the EFI partition we've likely already nuked the other partitions as well.

eslerm, 2023-07-28:
[from upstream](https://www.reddit.com/r/AsahiLinux/comments/159fg1a/psa_stop_telling_people_to_run_wipelinux/)

> ## PSA: Stop telling people to run wipe-linux
>
> That script is deliberately dangerous and only intended for developers wiping regularly.
>
> Going forward mentioning it [on reddit] is banned. Your comments/posts will be removed and repeat offenders will be banned. Yes, this is that serious.
> 
> If people are asking how to uninstall, send them to the [Partitioning cheatsheet](https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet) instead, which does mention that script with all the appropriate warnings that y'all never do, as well as documents the right way of doing things manually.

cowpod, 2023-07-28:
I fail to see how a reddit post contributes to this. We all know that the script can be dangerous. If you're somehow surprised that a script that warns it'll wipe everything... wipes everything, then I don't know what to say.