UbuntuAsahi/ubuntu-asahi #29: This device is not supported yet!

KpSoheilIstVergeben, 2023-05-09:
Collecting system information...
  Product name: MacBook Pro (14-inch, 2023)
  SoC: Apple M2 Pro
  Device class: j414sap
  Product type: Mac14,9
  Board ID: 0x4
  Chip ID: 0x6020
  System firmware: iBoot-8422.100.650
  Boot UUID: 6C075936-64B6-4683-86F4-089D79019B79
  Boot VGID: 6C075936-64B6-4683-86F4-089D79019B79
  Default boot VGID: 6C075936-64B6-4683-86F4-089D79019B79
  Boot mode: macOS
  OS version: 13.3.1 (22E261)
  SFR version: 22.5.261.0.0,0
  System rOS version: 13.3.1 (22E261)
  No Fallback rOS
  Login user: hauscheck

This device is not supported yet!
Please check out the Asahi Linux Blog for updates on device support:

   https://asahilinux.org/blog/

When will I be able to use this on my Mac?
Or is there any other way to install linux to my Mac?

tobhe, 2023-05-09:
Hi, as the error message says your device it not yet supported by Asahi (and by extension also not by ubuntu-asahi).

There is upstream work happening to support the M2 pro and minis, but new hardware means more reverse engineering, writing drivers and testing so the whole process usually takes a while depending on how different the new hardware is to the old one. 

From the public discussions my personal impression is that a lot of things are already working in an experimental state but might need some more fine tuning and polish before they can be released to the public. I expect that there will likely be a bigger Asahi release in the (possibly near) future that adds support for your M2, but I don't think anyone knows an exact date yet. Once that is out, I will update Ubuntu to ship the new packages which usually takes another week.

clementperon, 2023-06-18:
@KpSoheilIstVergeben I have the same laptop and I works fine on my machine. Did you retry ?

KpSoheilIstVergeben, 2023-06-18:
@clementperon Nah Im just using a Virtual Machine now

eldemaster, 2023-06-18:
@clementperon Hi mate do you feel any issues in running bare metal or everything's fine? I'm thinking of installing it but don't want to try and fail installation because it's not supported

tobhe, 2023-06-18:
To add some context: the latest kernel update added basic support for m2 pro,ultra and max devices according to https://social.treehouse.systems/@AsahiLinux/110497512340479064.

I have not uploaded updated install images with the new kernel yet because I was planning to see if I get any breakage reports first and I don't have a m2 pro machine here to test. If it works for @clementperon I guess we can go ahead and update them.

clementperon, 2023-06-18:
I'm running arch Linux on it for a long time but before you need to enable expert mode in the installer.

Biggest missing to me is Google Chrome that is not supported on arm64 (not related to Asahi).
Sound which needs BT speaker or headphones.

eldemaster, 2023-06-18:
@clementperon isn't Chromium available? I'm running it in a VM Ubuntu 22.04 with VMware Fusion currently.

clementperon, 2023-06-18:
Chromium is not chrome. You can run chromium if you want.

tobhe, 2023-06-19:
The 23.04 desktop installer image is now up-to-date and should work on m2 pro/edge/max. If anyone decides to give it a try, feel free to let us know how it went. As mentioned above I unfortunately don't have the hardware to test it myself.

eslerm, 2023-06-21:
@tobhe can we relabel this to `add virtual machine support`?

clementperon, 2023-06-21:
@eslerm why not just closing this issue and open another ? 

What do you mean by Virtual Machine you want Ubuntu asahi to support KVM ?

tobhe, 2023-06-21:
I would prefer to leave this one open until we have confirmation that the original issue is resolved.

Not sure if VM support is an issue, I don't see why that wouldn't work out of the box.

clementperon, 2023-06-21:
@eslerm 
```
$> ubuntu@localhost:~$ sudo virt-host-validate qemu
  QEMU: Checking if device /dev/kvm exists                                   : PASS
  QEMU: Checking if device /dev/kvm is accessible                            : PASS
  QEMU: Checking if device /dev/vhost-net exists                             : PASS
  QEMU: Checking if device /dev/net/tun exists                               : PASS
  QEMU: Checking for cgroup 'cpu' controller support                         : PASS
  QEMU: Checking for cgroup 'cpuacct' controller support                     : PASS
  QEMU: Checking for cgroup 'cpuset' controller support                      : PASS
  QEMU: Checking for cgroup 'memory' controller support                      : PASS
  QEMU: Checking for cgroup 'devices' controller support                     : PASS
  QEMU: Checking for cgroup 'blkio' controller support                       : PASS
  QEMU: Checking for device assignment IOMMU support                         : WARN (Unknown if this platform has IOMMU support)
  QEMU: Checking for secure guest support                                    : WARN (Unknown if this platform has Secure Guest support)
```

eslerm, 2023-06-21:
hi @clementperon, @KpSoheilIstVergeben opened this issue while having issues using a virtual machine, which is why I suggested the change. imo, this issue shouldn't apply to all device support

@tobhe I'm okay with keeping the issue open (unless vm is not-for-us)

tobhe, 2023-06-21:
@eslerm as I understand it they opened the issue because of missing m2 pro hardware support and are using a VM now as a workaround.
Maybe a better name would be "Add m2 pro/max/edge support"?

tobhe, 2023-07-14:
FWIW: Upstream Asahi is calling for testers for those machines here: https://social.treehouse.systems/@marcan/110712951601718261
I guess that could be a good way to help get them properly supported.

eslerm, 2023-10-29:
This is now supported.

closed 2023-10-29 by eslerm