UbuntuAsahi/ubuntu-asahi #30: sound driver

seyedanvari, 2023-05-17:
the sound driver not installed in ubuntu 22.10

seyedanvari, 2023-05-17:
I meant that the speaker was inactive. Thank you for your answer.

eslerm, 2023-06-21:
hi @seyedanvari

There is speaker support in the kernel, but it is not enabled for safety reasons. There can be hardware damage if the speakers are played too loud. Upstream is aware of this, and we will add support as soon as it is available.

https://github.com/AsahiLinux/docs/wiki/Feature-Support

PlusLake, 2023-11-19:
I can install speakersafetyd with Debian to enable speakers in my M1 Macbook Air since last week.
Can I do the same with ubuntu-asahi?

https://social.treehouse.systems/@AsahiLinux/111396388974903682

tobhe, 2023-11-19:
@PlusLake In theory you should be able to set up the userland parts (speakersafetyd, configs, etc) by hand to get working speaker support. We are working on packaging everything up nicely and enabling it by default and that unfortunately takes a little longer. We hope to have everything in place within the next week(s).

tobiasgrosser, 2023-12-20:
Speaker support should now be available upstream. However, the packages do not yet seem to work for me. In particular, I get the following error:

```
The following packages have unmet dependencies:
 asahi-audio : Depends: pipewire (>= 0.3.81) but 0.3.79-2 is to be installed
               Depends: wireplumber (>= 0.4.15-2~) but 0.4.14-4ubuntu1.asahi0+32~ubuntu23.10.1 is to be installed
               Recommends: lv2-bankstown but it is not installable
               Recommends: speakersafetyd but it is not installable
E: Unable to correct problems, you have held broken packages.
```

tobhe, 2023-12-21:
@tobiasgrosser yes, I am working on getting those packaged. bankstown is already uploaded to the asahi-testing ppa, speakersafetyd is WIP. For pipewire and wireplumber I am considering backporting the noble versions. The ones we have in mantic and especially jammy would need quite a few patches.

PlusLake, 2024-01-07:
perform a `sudo apt update && sudo apt upgrade` few minutes ago and speaker works great now!
Big thanks to all the contributors!! :heart:

tobiasgrosser, 2024-01-07:
This meanwhile works for me too. This issue is (partially?) addressed, I feel.

tobhe, 2024-01-07:
Speaker support should now be enabled by default on Mantic. Jammy needs a little more work to backport all the dependencies. Glad to hear that it works for you!

HasanTheSyrian, 2024-01-28:
> Speaker support should now be enabled by default on Mantic. Jammy needs a little more work to backport all the dependencies. Glad to hear that it works for you!

Do the speakers work on Jammy? I couldn't find anything related in the commits.

Thanks in advance.

eslerm, 2024-02-14:
Ubuntu Asahi has moved to 9 months of Ubuntu release support. Backporting speakers packages to Jammy was a factor. Please ask to re-open issue if a supported version is affected.

closed 2024-02-14 by eslerm