UbuntuAsahi/ubuntu-asahi #32: Error: -69716: Storage system verify or repair failed

termaulmaul, 2023-05-23:
Started APFS operation
Aligning shrink delta to 46,591,836,160 bytes and targeting a new container size of 198,515,359,744 bytes
Determined the minimum size for the APFS Container to be 164,651,597,824 bytes
Resizing APFS Container designated by APFS Container Reference disk3
The specific APFS Physical Store being resized is disk0s2
Verifying storage system
Using live mode
Performing fsck_apfs -n -x -l /dev/disk0s2
Checking the container superblock
Checking the EFI jumpstart record
Checking the space manager
Checking the space manager free queue trees
Checking the object map
Checking the encryption key structures
Checking volume /dev/rdisk3s1
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking snapshot 1 of 1 (com.apple.os.update-BC92420545C003D0A4FCC39C1820D7F53A30223F5B4CC3C4AA749692E7901C1E)
Checking the fsroot tree
Checking the file extent tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s1 appears to be OK
Checking volume /dev/rdisk3s2
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s2 appears to be OK
Checking volume /dev/rdisk3s3
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s3 appears to be OK
Checking volume /dev/rdisk3s4
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s4 appears to be OK
Checking volume /dev/rdisk3s5
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the document ID tree
Checking the fsroot tree
error: refcnt (1) of dstream id object (id 4476040) is less than expected (2)
error: doc-id tree: record exists for doc-id 165630, file-id 16417152 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 165672, file-id 16727906 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 173517, file-id 17131537 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 217080, file-id 19791093 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 229996, file-id 19824257 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 245625, file-id 19863314 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 281504, file-id 25786445 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 304233, file-id 27263843 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 304245, file-id 27264133 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 304311, file-id 27271499 but no inode references this doc-id
error: doc-id tree: record exists for doc-id 304320, file-id 27273137 but no inode references this doc-id
Checking the extent ref tree
Checking the file key rolling tree
Verifying volume object map space
The volume /dev/rdisk3s5 was found to be corrupt and needs to be repaired
Checking volume /dev/rdisk3s6
Checking the APFS volume superblock
Checking the object map
Checking the snapshot metadata tree
Checking the snapshot metadata
Checking the fsroot tree
Checking the extent ref tree
Verifying volume object map space
The volume /dev/rdisk3s6 appears to be OK
Verifying allocated space
Performing deferred repairs
error: refcnt of dstream id object (id 4476040) less than expected
error: doc-id tree record exists for doc-id 165630, but no inode references this doc-id
Skipped 11/11 repairs of this type in total
Deferred repairs skipped
The container /dev/disk0s2 could not be verified completely
Storage system check exit code is 8
Error: -69716: Storage system verify or repair failed

Resize failed. This is usually caused by pre-existing APFS filesystem corruption.
Carefully read the diskutil logs above for more information about the cause.
This can usually be solved by doing a First Aid repair from Disk Utility in Recovery Mode.

termaulmaul, 2023-05-23:
I've run First Aid in recovery mode on all the partitions, but when I run it again the results are the same. Please help, by the way this is a first-hand MacBook, I bought it at the official Indonesian store, and I've never had a clean install.

tobhe, 2023-05-23:
Hi @termaulmaul, thanks for the report.
I don't recognize the error message you posted. Could you give a little more context on what you were doing to trigger the bug and what exact hardware you are using? I am assuming this error happened while running the installer. Is your machine still booting?

termaulmaul, 2023-05-24:
> Hi @termaulmaul, thanks for the report. I don't recognize the error message you posted. Could you give a little more context on what you were doing to trigger the bug and what exact hardware you are using? I am assuming this error happened while running the installer. Is your machine still booting?

yes my machine still booting, and error when resizing install on command line

tobhe, 2023-05-24:
Which means this is likely an APFS bug or hardware failure as the error message suggest and not actually a bug in the Asahi installer. Can you manually resize your APFS partition with `Disk Utility` or with `diskutil` on the command line?
How much did you plan to shrink it?

charlesalary, 2023-09-13:
I was updating to Monterey through OCLP and during the last portion I got 'storage system verify or repair failed (-69716)'
I've tried all the web searches suggestions and I'm stuck. I got two other computers to install OK but my laptop has this particular problem.
Any help would be appreciated.

rarmellini, 2024-12-10:
> I was updating to Monterey through OCLP and during the last portion I got 'storage system verify or repair failed (-69716)' I've tried all the web searches suggestions and I'm stuck. I got two other computers to install OK but my laptop has this particular problem. Any help would be appreciated.

Did you get your issue solved? I am getting the same message when installing MacOS Sonoma on as iMac Late 2015

Vladimir-Moskalenko, 2024-12-10:
I may not be an expert here, but isn't Ubuntu Asahi only for Apple Silicon Macs? OCLP is a tool to let Intel Macs run newer macOS versions, which implies you're using an Intel Mac.