UbuntuAsahi/ubuntu-asahi #33: GPU drivers

HeshamAbdelgawad, 2023-05-24:
When will the gpu drivers for ubuntu-asahi be released?

tobhe, 2023-05-24:
Hi @HeshamAbdelgawad. GPU drivers are supported in kinetic and newer. Which version are you using?

HeshamAbdelgawad, 2023-05-24:
I am using lunar

tobhe, 2023-05-24:
Is this a fresh install or updated from a previous release? Can you share what `cat /etc/apt/sources.list.d/asahi-ppa.list` and `apt list --installed | grep asahi` return for you?

HeshamAbdelgawad, 2023-05-24:
[image]

tobhe, 2023-05-24:
so kernel and mesa are there and no `/etc/apt/sources.list.d/asahi-ppa.list` means this is a relatively new installation. Try `cat /etc/apt/sources.list.d/extra-ppas.list` instead but since this is a new installation I guess this shouldn't be the issue.

Why do you think the GPU is not working? What does `glxinfo | grep -C 4 Accelerated` give you (you might need to install `mesa-utils` for that)?

HeshamAbdelgawad, 2023-05-24:
It says next to acceleration yes.
15 minutes ago, when i executed "screenfetch", the gpu section was empty, but now it recognized "apple m1".
thanks for your help.

tobhe, 2023-05-24:
Weird, but I'm glad it works now :) Closing this bug for now. Feel free to reopen this if it happens again.

closed 2023-05-24 by HeshamAbdelgawad