UbuntuAsahi/ubuntu-asahi #34: dropbear-initramfs doesn't run in 22.04 server

vercas, 2023-05-26:
The title says it all, pretty much.  
The package installs. `update-initramfs` then complains that I have no authorized_keys file for Dropbear, meaning that at least some of its initramfs hooks work.  
I add my keys there, run `update-initramfs -u -k all` and `update-grub` just for good measure.  
I reboot, but Dropbear doesn't seem to start...  
I've added an encrypted disk with the `initramfs` option, hoping to be able to unlock it via Dropbear.  
Ubuntu leases an IP address via DHCP and responds to pings while waiting for the disk to be decrypted.  

I've used the same setup on many other Ubuntu 22.04 and 20.04 machines, and it works fine on them.

tobhe, 2023-06-27:
I have managed to reproduce the issue on my M1 mini and I found a way to get it working. The first thing I had to do was making the tg3 module available in the initramfs. For that, you simply need to add tg3 to `/etc/initramfs-tools/modules`.

The second one is a bit trickier: There seems to be a bug in the kernel module causing it to not properly declare its firmware dependency. I manually copied the firmware file to my initrd with a custom hook:

```
ubuntu@ubuntu:~$ cat /etc/initramfs-tools/hooks/tg3
#!/bin/sh -e
# Copy the compatibility symlinks until initramfs-tools will be converted
# to use the kmod program.

if [ "$1" = "prereqs" ]; then exit 0; fi

. /usr/share/initramfs-tools/hook-functions

mkdir -p "$DESTDIR/lib/firmware/tigon"
cp -a /lib/firmware/tigon/tg357766.bin "$DESTDIR/lib/firmware/tigon"
```

With that dropbear works for me. I am working on a proper kernel fix for the latter.

tobhe, 2023-06-28:
I submitted the kernel fix here: https://lore.kernel.org/lkml/ZJt7LKzjdz8+dClx@tobhe.de/T/#u
With that we don't need the custom hooks. I will also include the fix in the future ubuntu-asahi kernel uploads.

tobhe, 2023-06-28:
interesting. So I tested by passing the `break` option on the kernel command line and managed to log in successfully, I haven't actually set up LUKS but that should be equivalent (or so i thought). You could try using break and see if you find out what's wrong. Another thing i forgot to mention is the initrd network configuration: I had to add `IP=::::ubuntu::dhcp` to `/etc/initramfs-tools/initramfs.conf`

vercas, 2023-07-28:
```
❯ lsinitramfs /boot/initrd.img-6.1.0-1-asahi | grep tg3
usr/lib/firmware/tigon/tg3.bin
usr/lib/firmware/tigon/tg357766.bin
usr/lib/firmware/tigon/tg3_tso.bin
usr/lib/firmware/tigon/tg3_tso5.bin
usr/lib/modules/6.1.0-1-asahi/kernel/drivers/net/ethernet/broadcom/tg3.ko
```

Also added the line you've mentioned there.

Still doesn't work.

tobhe, 2023-07-28:
hm maybe there is an ordering problem. You should be able to control when your want to break out of the initramfs. Try picking an earlier one

eslerm, 2024-02-14:
Ubuntu Asahi has moved to 9 months of Ubuntu release support. Please ask to re-open if a current version is affected. (Currently server isn't part of our installer, but might be soon.)

closed 2024-02-14 by eslerm

vercas, 2024-08-05:
I've dusted off my Mac Mini and tried again after updating it.  
It's working perfectly now. 😃