UbuntuAsahi/ubuntu-asahi #38: Successful 22.04 install -- Can't login after restart

Chambana, 2023-07-02:
Using a Macbook Air M1.  Installed using the prebuilt image and install script.  System was stable after a full day of use.  Booted the system this morning and I can't get a login prompt. 

[image]
[image]

eslerm, 2023-07-02:
Thank you for the report.

After the initial install, were you able to successfully reboot the device?

Chambana, 2023-07-02:
Yes I was. I had a successful reboot iirc.

Chambana, 2023-07-02:
Are there are logs or troubleshooting I can do to help diagnose the issue?

Chambana, 2023-07-02:
As far as changes made the machine after install, I did at least the following:

- installed firefox via apt
- installed chromium-browser via apt
- installed ROS2 Humble via apt
- etc

tobhe, 2023-07-02:
Can you switch to a tty with ctrl-alt-fn-f3? If yes, can you show us the output of `apt list --installed | grep asahi`? Another think that might work is holding ESC after the u-boot automatic boot timeout or ctrl-alt-del to get a grub boot menu and try if you can boot an older kernel (if there is one).

Chambana, 2023-07-02:
Yes, I'll log my results here:

- `fn-ctrl-option-f3 `successfully dropped me to a tty login prompt
- I've attached the results of `apt list --installed | grep asahi`. 
- [image]
- tapping `esc` right after the autoboot timeout did drop me a grub menu w/2 available kernels -- **6.1.0-1asahi** and **5.19.5-10-asahi**.
- [image]
- From a cold boot, I've confirmed that the **5.19.5-10-asahi** kernel boots as expected into a desktop login prompt and I can successfully login and drop to a desktop environment.  This is the login screen that I land on before entering my password
[image]

- From a cold boot, I've confirmed that the **6.1.0-1-asahi** kernel drops me to a deadend login screen with no login prompt
[image]

Is the 6.1.0-1-asahi supported on a macbook air m1?

Chambana, 2023-07-02:
I used the Step 7 of this grub [tutorial](https://linux.how2shout.com/how-to-change-default-kernel-in-ubuntu-22-04-20-04-lts/) to set my default boot kernel to 5.19.5-10-asahi and the system boots consistently.

tobhe, 2023-07-02:
ok, so it looks like our kernel is broken on your hardware and we need to roll out an update to fix it.

Here's how you could help us with that if you have some spare time:
1. I have already built a newer kernel here https://launchpad.net/~tobhe/+archive/ubuntu/asahi-kernel-build/+sourcepub/14980499/+listing-archive-extra
2. You can download it by clicking linux-image-6.3.0-7-asahi_6.3.0-7.8_arm64.deb and linux-modules-6.3.0-7-asahi_6.3.0-7.8_arm64.deb
3. Next `apt install ./linux-image-6.3.0-7-asahi_6.3.0-7.8_arm64.deb ./linux-modules-6.3.0-7-asahi_6.3.0-7.8_arm64.deb`, that should automatically trigger a grub update adding a new boot entry. 
4. See if that new kernel works. If it does we can release it

Chambana, 2023-07-02:
Happy to.  
I followed the steps above and manually installed the 6.3.0-7.8 kernel. I manually selected this new kernel from the grub menu (extra step because i forced grub earlier to stick with 5.19).  The new kernel successfully dropped me to a desktop GUI as intended and this behavior survived several reboots.  So, anecdotally, this kernel update totally fixed my issue. Thanks! 
(it also resolved a screen brightness issue I was having on 5.19, but that would've been a separate ticket.  Unfortunately, it didn't resolve the disgustingly dirty state of my computer screen and I'll have to manually handle that with some windex. I swear it doesn't look that dirty in real life).    

[image]
[image]
[image]

closed 2023-07-02 by Chambana

Chambana, 2023-07-03:
FWIW, I also see a 30% increase in aggregate CPU performance when moving from the 5.19 kernel above to the new 6.3 kernel.

**6.3 kernel results**
```
 PassMark PerformanceTest Linux

Apple MacBook Air (M1, 2020)
 (aarch64)
4 cores @ 0 MHz  |  15.3 GiB RAM
Number of Processes: 8  |  Test Iterations: 1  |  Test Duration: Medium
--------------------------------------------------------------------------
CPU Mark:                          5915
  Integer Math                     26489 Million Operations/s
  Floating Point Math              36337 Million Operations/s
  Prime Numbers                    56.0 Million Primes/s
  Sorting                          20511 Thousand Strings/s
  Encryption                       1244 MB/s
  Compression                      39847 KB/s
  CPU Single Threaded              1833 Million Operations/s
  Physics                          1031 Frames/s
  Extended Instructions (NEON)     7683 Million Matrices/s

Memory Mark:                       2664
  Database Operations              3686 Thousand Operations/s
  Memory Read Cached               14766 MB/s
  Memory Read Uncached             14859 MB/s
  Memory Write                     14916 MB/s
  Available RAM                    13337 Megabytes
  Memory Latency                   32 Nanoseconds
  Memory Threaded                  43345 MB/s
--------------------------------------------------------------------------
```

**6.3 kernel results**
   ```
 PassMark PerformanceTest Linux

Apple MacBook Air (M1, 2020)
 (aarch64)
4 cores @ 2064 MHz  |  15.3 GiB RAM
Number of Processes: 8  |  Test Iterations: 1  |  Test Duration: Medium
--------------------------------------------------------------------------
CPU Mark:                          7640
  Integer Math                     32196 Million Operations/s
  Floating Point Math              50176 Million Operations/s
  Prime Numbers                    71.3 Million Primes/s
  Sorting                          26236 Thousand Strings/s
  Encryption                       1680 MB/s
  Compression                      52864 KB/s
  CPU Single Threaded              2941 Million Operations/s
  Physics                          1174 Frames/s
  Extended Instructions (NEON)     7308 Million Matrices/s

Memory Mark:                       3595
  Database Operations              4835 Thousand Operations/s
  Memory Read Cached               23849 MB/s
  Memory Read Uncached             23714 MB/s
  Memory Write                     23934 MB/s
  Available RAM                    13554 Megabytes
  Memory Latency                   28 Nanoseconds
  Memory Threaded                  43372 MB/s
--------------------------------------------------------------------------

```

tobhe, 2023-07-03:
Now that's an interesting data point. I am not sure what causes this but it might be some kind of cpu power management improvement. More reasons to push out the update I guess.

Chambana, 2023-07-03:
Results from running the same benchmark on the same hardware on native MacOS.  For some reason the Ubuntu 6.3 kernel is only showing 4 cores (not 8) and a much lower clock rate for the CPU.  

```

                          PassMark PerformanceTest Mac

MacBookAir10,1
Apple M1 (arm64)
8 cores @ 3200 MHz  |  16.0 GiB RAM
Number of Processes: 8  |  Test Iterations: 1  |  Test Duration: Medium
--------------------------------------------------------------------------
CPU Mark:                          14478
  Integer Math                     32766 Million Operations/s
  Floating Point Math              37261 Million Operations/s
  Prime Numbers                    168 Million Primes/s
  Sorting                          22781 Thousand Strings/s
  Encryption                       9053 MB/s
  Compression                      177726 KB/s
  CPU Single Threaded              3820 Million Operations/s
  Physics                          1367 Frames/s
  Extended Instructions (NEON)     6320 Million Matrices/s

Memory Mark:                       3926
  Database Operations              6804 Thousand Operations/s
  Memory Read Cached               23855 MB/s
  Memory Read Uncached             23802 MB/s
  Memory Write                     24055 MB/s
  Available RAM                    10868 Megabytes
  Memory Latency                   22 Nanoseconds
  Memory Threaded                  57788 MB/s
--------------------------------------------------------------------------

```

tobhe, 2023-07-03:
I think there might be a little more gain once all packages are up-to-date, not just the kernel. Not sure how passmark determines the number of cores but try running `nproc` or `htop` and see what that says.

space192, 2023-07-04:
Hey ! So I followed the same steps from one you previous answer, in this thread I also have everything working but the GPU acceleration ! as it's linked to kernel 6.1