UbuntuAsahi/ubuntu-asahi #40: Excessive minimum size required for partition

ilias-t, 2023-07-03:
The installer is enforcing a surprisingly large amount of space as the minimum for the partition size. Why would it ever need +600GBs of size?

[image]

Notes:
- Time Machine is not setup.
- Mac OS is 14.0 Beta

eslerm, 2023-07-03:
Hi @ilias-t o/

What model of mac do you have? Have you ran the installer more than once and seen this? Do you get the same result running upstream's install script https://github.com/AsahiLinux/asahi-installer ?

ilias-t, 2023-07-03:
Hey @eslerm I'm on an M1. I also just tried running the bootstrap script in the upstream repo and was able to reproduce the same result. Let me know if you'd prefer if I move the issue there instead.

eslerm, 2023-07-03:
Thanks! 

I haven't come across `overhead space` before, but that sounds like the issue. Could you run `diskutil list` in MacOS and share the results?

The system should look like the before in https://github.com/UbuntuAsahi/ubuntu-asahi/issues/28#issuecomment-1539568477 [More info in upstream's Partitioning Cheatsheet](https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet#listing-partitions-with-diskutil).

ilias-t, 2023-07-04:
Sure thing, here are the results from `diskutil list`:

[image]

eslerm, 2023-07-04:
Compared to the initial config [in this video](https://www.youtube.com/watch?v=aXh_d1OixRI), the partition names and order differ. [image]

The man page for `diskutil` hints at how "overhead" occurs in the `resizeVolume` section:
> You can grow a volume (back) to its maximum size possible, provided no new partitions have been created that are in the way, by specifying R for the new volume size. You should use R instead of attempting an absolute value such as 100% because the latter cannot count partition map overhead.

ilias-t, 2023-07-04:
Hmm, I'm not seeing the difference. Here's another shot without my external USB plugged in that looks mostly identical. Not sure what else to try given there are no time machine backups. I also tried running First Aid on the disk and it came back fine. I wonder if it's related to being on the beta OS.

[image]

ilias-t, 2023-07-04:
Is there anyway to instruct the script to use an already existing partition?

eslerm, 2023-07-04:
In the other example `Machintosh HD - Data` is `disk3s1` and there's some reordering.

The [rest of the error code](https://github.com/AsahiLinux/asahi-installer/blob/04c037af6e58fea856003882a06e64ad100c81cc/src/main.py#L728) suggests resizing the volume. Possibly `diskutil resizeVolume disk0s2 R` ?
 
The most certain test is to erase and reinstall macos using recovery. *If* you do, could you please try running the installer partially before and after updating to 14.0 to find the suggested `min_size` for each?

ilias-t, 2023-07-04:
Reinstalled the OS and no luck. Going to close this as it doesn't seem to be an issue with this project. Thanks for help!

closed 2023-07-04 by ilias-t

eslerm, 2023-07-04:
Thanks for checking reinstall. Upstream will definitely be interested if this is MacOS 14.0 specific. If you file an issue, please link it.

eslerm, 2023-07-15:
There was a MacOS 14.0 (Sonoma) report upstream (https://github.com/AsahiLinux/asahi-installer/issues/199) but, in the logs there is zero overhead. So, it feems unlikely that MacOS 14.0 itself is causing overhead partitioning.

marcan, 2023-07-16:
The new size is the new size for your macOS partition. You have ~1TB total, 431GB free space, and 382GB available space after purgeable file overhead. That makes your minimum macOS partition size around 600GB. Everything is working as intended here.

The extra 11GB overhead is probably because you have a stale installer for Sonoma or something like that left over. Unless you're worried about squeezing down your macOS install to the minimum size (which I doubt you intend, since you're using more than half of your disk on it already), there is nothing to worry about.

ilias-t, 2023-07-16:
@marcan oh, thank you for clarifying! I was under the impression that it was needing the 600GB for the new partition size. It makes perfect sense now that it's asking how much space to allocate to the macOS partition. Much appreciated