UbuntuAsahi/ubuntu-asahi #41: Grub Rescue prompt is shown and computer doesn't boot into OS.

SamilKaraduman, 2023-07-05:
[image]

Ever since I reinstalled Ubuntu on my computer, it booted only once into Ubuntu (first setup) and every time after that refused to boot. Uboot, (or rather grub) displays a grub rescue prompt looking like the one in the picture. How can I resolve this? 

OS: Ubuntu 23.04 Lunar Lobster
Host: MBA (M1, 2020)

Any help is heavily appreciated.

tobhe, 2023-07-05:
Hi @SamilKaraduman, there was a known bug that caused similar problems, but I thought that was fixed. Did you reinstall recently?

SamilKaraduman, 2023-07-05:
Yes, I have reinstalled recently. About a day ago to be precise.

tobhe, 2023-07-05:
Sounds like bad news. Last time it was caused by a race in the first-boot script, so it wasn't triggered reliably. You could try reinstalling and see if first-boot.service succeeds before rebooting the machine. I will try to replicate it on my test machine, but that will take a few days.

SamilKaraduman, 2023-07-05:
Sure. I can do that. But I also want to clear boot policies on my drive, since too many can also cause problems. But how would I go about doing that?

tobhe, 2023-07-05:
I usually start into recovery mode, open a terminal there and run `curl -L https://alx.sh/cleanbp | sh` to clean the boot policies. Watch out though because I think that nukes everything but macos.

SamilKaraduman, 2023-07-05:
Cleaned the boot policies. The script worked just fine, and I rebooted into macOS. I then reinstalled Ubuntu (before the script I already removed Ubuntu from my computer) and everything went fine, I booted into Ubuntu, got everything setup until the next boot... Grub/Uboot still gave me the grub rescue screen. What do I do now? I haven't poked around with anything and now I am unable to boot into Ubuntu.

tobhe, 2023-07-05:
ok, so what we need is the actual error message. I am assuming what's failing is the first-boot script, so we need the output of `sudo journalctl -eu first-boot` after a fresh reinstall. I can't reproduce it on my machine so I will need some help here.

SamilKaraduman, 2023-07-05:
So I'm gonna need to do a 3rd install?!?!

tobhe, 2023-07-05:
Not necessarily. The alternative would be booting an usb image like https://files.tobhe.de/ubuntu/ubuntu-desktop-live-23.04-20230329.img by flashing it to a usb drive and typing `run bootcmd_usb0` into the u-boot prompt, then mounting the ubuntu filesystems, chrooting into the mounting system and fixing it from there.

I think running `sudo update-grub` might be enough to fix it, but I can only guess with the little info I have right now.

SamilKaraduman, 2023-07-06:
[image]

Used dd to flash the image onto a spare flash drive of mine, and it also gives a grub rescue boot menu. Perhaps something is corrupted with the latest images?

tobhe, 2023-07-06:
urgh no this actually makes a lot of sense. the usb is build from the same script as the regular install so their filesystem UUIDs collide (further hardening the suspicion that this is a first-boot bug). Another usb image such as https://arvanta.net/alpine/install-alpine-m1/ should work.

SamilKaraduman, 2023-07-06:
*sigh* Tried to boot Alpine, and I still got a grub rescue error! I then tried booting Fedora, another grub rescue prompt! Something is definitely very wrong here...

dheerajgattupalli, 2023-07-07:
I am facing similar issue, I am working with a bluetooth keyboard so actually stuck at grub rescue... @SamilKaraduman did you try the update-grub command?

tobhe, 2023-07-07:
@dheerajgattupalli and this also happened after a fresh 23.04 install's first reboot? Would you mind trying to reproduce it and BEFORE you configure everything and restart the system share your systemd journal and in particular `sudo journalctl -u first-boot`?

dheerajgattupalli, 2023-07-07:
[log_firstboot.log](https://github.com/UbuntuAsahi/ubuntu-asahi/files/11979201/log_firstboot.log)

here is the log, didnt restart yet.. let me know if you need anything else

tobhe, 2023-07-07:
Thank you @dheerajgattupalli! I don't see any obvious errors there. A few more things that would help: `lsblk --fs /dev/nvme0n1` and the contents of `/boot/grub/grub.cfg` to see if the file system UUIDs in the config match what's actually there.

dheerajgattupalli, 2023-07-07:
**lsblk --fs /dev/nvme0n1 output:**

NAME        FSTYPE FSVER LABEL       UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
nvme0n1                                                                                  
├─nvme0n1p1 apfs                     f5e60141-983a-45d1-a1a4-bfde1824a10d                
├─nvme0n1p2 apfs                     f453a203-6f62-4547-9ae8-171b86f27544                
├─nvme0n1p3 apfs                     62af437f-abd4-4fa2-880f-79334ae699c4                
├─nvme0n1p4 vfat   FAT32 EFI - UBUNT F73D-093D                             404.1M    15% /boot/efi
├─nvme0n1p5 ext4   1.0   ubuntu-root aebcdc29-7122-4f09-aee3-f8edd81e45f3     30G    18% /var/snap/firefox/common/host-hunspell
│                                                                                        /
└─nvme0n1p6 apfs                     bc59d466-ddd6-436f-bd81-83c375fefddc         

[grub.txt](https://github.com/UbuntuAsahi/ubuntu-asahi/files/11979405/grub.txt)
and this is the grub.cfg

tobhe, 2023-07-07:
Thanks, this also looks correct. We can see that your root UUID is `aebcdc29-7122-4f09-aee3-f8edd81e45f3` which matches the boot menu entry which sets `root=UUID=aebcdc29-7122-4f09-aee3-f8edd81e45f3` which again matches what we see in the first-boot log. Your installation looks fine to me.

dheerajgattupalli, 2023-07-07:
Thanks for your help, just restarted i didnt get the error this time. sorry, so those logs may not be useful.. first time i applied ubuntu updates before restart so maybe those messed up my installation. is it ok to apply software updates? one of the update was called "ubuntu-base"

tobhe, 2023-07-07:
The fact that the bug is not reliably reproducible even on the same hardware is already valuable information, thanks again for your help @dheerajgattupalli! Software updates absolutely should be safe to apply.

SamilKaraduman, 2023-07-07:
As for me, what should I go about doing now? Seeing that I can't even boot off of a flash drive theres not much I can do. Has there been any updates pushed out yet? I'm willing to do a fresh install if everything is fixed, unless there is another way.

tobhe, 2023-07-07:
@SamilKaraduman No updates yet since we don't know what's happening. We need someone to share a log of the breakage to figure out what's going on. Unfortunately I can't really think of another way other than booting from USB which also seems to be broken.

If you just want it to work I propose the following:
1. reinstall
2. DON'T REBOOT
3. check/share your logs to see what's wrong
4. fix it by hand (if you don't know how, I can certainly help)
5. have a working system

Maybe if you are lucky we can even skip some points if it magically works this time.

SamilKaraduman, 2023-07-07:
Guess I'll be doing a full reinstallation then, will take about an hour or so. I'll post an update when I'm done.

SamilKaraduman, 2023-07-07:
System is up and running. I am currently running Ubuntu while typing this, and this is the output of the command:

`Jul 07 20:42:34 localhost.localdomain systemd[1]: Starting first-boot.service ->
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + set -e
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + trap cleanup EXIT
Jul 07 20:42:34 localhost.localdomain first-boot[584]: + findmnt -n -o UUID /
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + root_uuid_old=87c6b0ce>
Jul 07 20:42:34 localhost.localdomain first-boot[587]: + findmnt -n -o SOURCE /
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + root_dev=/dev/disk/by->
Jul 07 20:42:34 localhost.localdomain first-boot[590]: + findmnt -n -o SOURCE />
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + efi_dev=/dev/nvme0n1p4
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/disk/by-uuid>
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + echo Randomizing root >
Jul 07 20:42:34 localhost.localdomain first-boot[583]: Randomizing root filesys>
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + udevadm control -s
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + tune2fs -U random /dev>
Jul 07 20:42:34 localhost.localdomain first-boot[592]: tune2fs 1.47.0 (5-Feb-20>
Jul 07 20:42:34 localhost.localdomain first-boot[599]: + grep ^UUID=
Jul 07 20:42:34 localhost.localdomain first-boot[598]: + blkid -c /dev/null /de>
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + root_uuid=UUID=c32f8a9>
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + udevadm control -S
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + echo Root filesystem: >
Jul 07 20:42:34 localhost.localdomain first-boot[583]: Root filesystem: UUID=c3>
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + echo
Jul 07 20:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/nvme0n1p4 ]
lines 1-23

`

SamilKaraduman, 2023-07-07:
As for fixing the system, I would like some help on that. (Please!)

tobhe, 2023-07-08:
@SamilKaraduman that log doesn't look complete and we need to find out why, if you look at `log_firstboot.log` above you can see that there are a few more commands after that.  Did it hang up or did you not copy all of it?

As above: what do `lsblk --fs /dev/nvme0n1` and `/boot/grub/grub.cfg` say about the uuid? it should be a new random one now and both outputs should match.

SamilKaraduman, 2023-07-08:
If i paste the above command, it only displays lines 1-23 in the terminal. If i resize the window the same output is outputted a couple of times, all saying the same thing. Where is the file even located? I'm currently up at 5am in my timezone trying to solve this issue.

SamilKaraduman, 2023-07-08:
Never mind, if i resize the window there seems to be a bit more than I thought.

`Jul 07 14:42:34 localhost.localdomain systemd[1]: Starting first-boot.service ->
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + set -e
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + trap cleanup EXIT
Jul 07 14:42:34 localhost.localdomain first-boot[584]: + findmnt -n -o UUID /
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_uuid_old=87c6b0ce>
Jul 07 14:42:34 localhost.localdomain first-boot[587]: + findmnt -n -o SOURCE /
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_dev=/dev/disk/by->
Jul 07 14:42:34 localhost.localdomain first-boot[590]: + findmnt -n -o SOURCE />
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + efi_dev=/dev/nvme0n1p4
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/disk/by-uuid>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo Randomizing root >
Jul 07 14:42:34 localhost.localdomain first-boot[583]: Randomizing root filesys>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -s
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + tune2fs -U random /dev>
Jul 07 14:42:34 localhost.localdomain first-boot[592]: tune2fs 1.47.0 (5-Feb-20>
Jul 07 14:42:34 localhost.localdomain first-boot[599]: + grep ^UUID=
Jul 07 14:42:34 localhost.localdomain first-boot[598]: + blkid -c /dev/null /de>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_uuid=UUID=c32f8a9>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -S
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo Root filesystem: >
Jul 07 14:42:34 localhost.localdomain first-boot[583]: Root filesystem: UUID=c3>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/nvme0n1p4 ]
lines 1-23...skipping...
Jul 07 14:42:34 localhost.localdomain systemd[1]: Starting first-boot.service - Sets up system on first boot...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + set -e
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + trap cleanup EXIT
Jul 07 14:42:34 localhost.localdomain first-boot[584]: + findmnt -n -o UUID /
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_uuid_old=87c6b0ce-3bb6-4dc2-9298-3a799bbb5994
Jul 07 14:42:34 localhost.localdomain first-boot[587]: + findmnt -n -o SOURCE /
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_dev=/dev/disk/by-uuid/87c6b0ce-3bb6-4dc2-9298-3a799bbb5994
Jul 07 14:42:34 localhost.localdomain first-boot[590]: + findmnt -n -o SOURCE /boot/efi
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + efi_dev=/dev/nvme0n1p4
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/disk/by-uuid/87c6b0ce-3bb6-4dc2-9298-3a799bbb5994 ]
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo Randomizing root filesystem UUID...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: Randomizing root filesystem UUID...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -s
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + tune2fs -U random /dev/disk/by-uuid/87c6b0ce-3bb6-4dc2-9298-3a799b>
Jul 07 14:42:34 localhost.localdomain first-boot[592]: tune2fs 1.47.0 (5-Feb-2023)
Jul 07 14:42:34 localhost.localdomain first-boot[599]: + grep ^UUID=
Jul 07 14:42:34 localhost.localdomain first-boot[598]: + blkid -c /dev/null /dev/disk/by-uuid/87c6b0ce-3bb6-4dc2-9298-3a799>
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + root_uuid=UUID=c32f8a9e-9c43-4d9b-9322-20804d799f03
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -S
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo Root filesystem: UUID=c32f8a9e-9c43-4d9b-9322-20804d799f03
Jul 07 14:42:34 localhost.localdomain first-boot[583]: Root filesystem: UUID=c32f8a9e-9c43-4d9b-9322-20804d799f03
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + [ -e /dev/nvme0n1p4 ]
Jul 07 14:42:34 localhost.localdomain first-boot[620]: + grep -q TYPE="vfat"
Jul 07 14:42:34 localhost.localdomain first-boot[619]: + blkid /dev/nvme0n1p4
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + echo Randomizing EFI system partition UUID...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: Randomizing EFI system partition UUID...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -s
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + fatlabel -i -r /dev/nvme0n1p4
Jul 07 14:42:34 localhost.localdomain first-boot[639]: There are differences between boot sector and its backup.
Jul 07 14:42:34 localhost.localdomain first-boot[639]: This is mostly harmless. Differences: (offset:original/backup)
Jul 07 14:42:34 localhost.localdomain first-boot[639]:   65:01/00
Jul 07 14:42:34 localhost.localdomain first-boot[639]:   Not automatically fixing this.
Jul 07 14:42:34 localhost.localdomain first-boot[644]: + blkid -c /dev/null /dev/nvme0n1p4 -o export
Jul 07 14:42:34 localhost.localdomain first-boot[645]: + grep ^UUID=
lines 1-35

Jul 07 14:42:34 localhost.localdomain systemd[1]: Starting first-boot.service - Sets up system on first boot...
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + set -e
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + trap cleanup EXIT
Jul 07 14:42:34 localhost.localdomain first-boot[584]: + findmnt -n -o UUID /
[... cut ...]
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + udevadm control -s
Jul 07 14:42:34 localhost.localdomain first-boot[583]: + fatlabel -i -r /dev/nvme0n1p4
Jul 07 14:42:34 localhost.localdomain first-boot[639]: There are differences between boot sector and its backup.
Jul 07 14:42:34 localhost.localdomain first-boot[639]: This is mostly harmless. Differences: (offset:original/backup)
Jul 07 14:42:34 localhost.localdomain first-boot[639]:   65:01/00
Jul 07 14:42:34 localhost.localdomain first-boot[639]:   Not automatically fixing this.
lines 1-33

`

tobhe, 2023-07-08:
The easiest way to make it readable would be piping it to a file and uploading that somewhere, so `sudo journalctl -u first-boot > first-boot.log`. I still don't see the service finishing in the log snippet you provided.

What do `lsblk --fs /dev/nvme0n1` and `cat /boot/grub/grub.cfg` say?

SamilKaraduman, 2023-07-08:
[first-boot.log](https://github.com/UbuntuAsahi/ubuntu-asahi/files/11990340/first-boot.log)

My bad, here it is. 

As for the other two commands, the lsblk command linked above gives me this output:

`NAME FSTYPE FSVER LABEL UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
nvme0n1
                                                                            
├─nvme0n1p1
│    apfs               e9f44fd6-b2f5-40dd-b292-01dbbe9cc699                
├─nvme0n1p2
│    apfs               d29b84d7-7719-4303-ad69-5ff2e355f406                
├─nvme0n1p3
│    apfs               10f0b933-78b5-4ffa-a0b7-a49e1ce92859                
├─nvme0n1p4
│    vfat   FAT32 EFI - UBUNT
│                       C9AC-CB8E                               404M    15% /boot/efi
├─nvme0n1p5
│    ext4   1.0   ubuntu-root
│                       c32f8a9e-9c43-4d9b-9322-20804d799f03   87.5G     8% /var/snap/firefox/common/host-hunspell
│                                                                           /
└─nvme0n1p6
     apfs               6467b3b6-c797-46c5-9c9b-cb109c475262                
`

And the cat command will give this output: 

`#
# DO NOT EDIT THIS FILE
#
# It is automatically generated by grub-mkconfig using templates
# from /etc/grub.d and settings from /etc/default/grub
#

### BEGIN /etc/grub.d/00_header ###
if [ -s $prefix/grubenv ]; then
  set have_grubenv=true
  load_env
fi
if [ "${initrdfail}" = 2 ]; then
   set initrdfail=
elif [ "${initrdfail}" = 1 ]; then
   set next_entry="${prev_entry}"
   set prev_entry=
   save_env prev_entry
   if [ "${next_entry}" ]; then
      set initrdfail=2
   fi
fi
if [ "${next_entry}" ] ; then
   set default="${next_entry}"
   set next_entry=
   save_env next_entry
   set boot_once=true
else
   set default="0"
fi

if [ x"${feature_menuentry_id}" = xy ]; then
  menuentry_id_option="--id"
else
  menuentry_id_option=""
fi

export menuentry_id_option

if [ "${prev_saved_entry}" ]; then
  set saved_entry="${prev_saved_entry}"
  save_env saved_entry
[... 153 lines cut ...]

### BEGIN /etc/grub.d/20_linux_xen ###

### END /etc/grub.d/20_linux_xen ###

### BEGIN /etc/grub.d/30_os-prober ###
### END /etc/grub.d/30_os-prober ###

### BEGIN /etc/grub.d/35_fwupd ###
### END /etc/grub.d/35_fwupd ###

### BEGIN /etc/grub.d/40_custom ###
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
### END /etc/grub.d/40_custom ###

### BEGIN /etc/grub.d/41_custom ###
if [ -f  ${config_directory}/custom.cfg ]; then
  source ${config_directory}/custom.cfg
elif [ -z "${config_directory}" -a -f  $prefix/custom.cfg ]; then
  source $prefix/custom.cfg
fi
### END /etc/grub.d/41_custom ###
`

tobhe, 2023-07-08:
Thank you, so the job did finish and it looks like the uuids are all correct this time. `c32f8a9e-9c43-4d9b-9322-20804d799f03` is the new random uuid and the log and grub config suggest it is in sync.

It will probably just work after reboot this time.

SamilKaraduman, 2023-07-08:
> EDIT: But I would suggest trying it before setting up all your configs

Try to reboot first?

tobhe, 2023-07-08:
yes

SamilKaraduman, 2023-07-08:
Finally, it works! Thank you so very much for all your time! After all this, do you think it is safe to update in the future? Definitely don't want something like this to happen again!

tobhe, 2023-07-08:
yes it should be safe to update. I take good care to test all updates but of course sometimes bugs happen that are related to a specific hardware/software combination. If that happens I am of course reading the reports here and try to get them fixed.

tyleraiv, 2023-07-19:
Apparently I'm experiencing the same issue.  I've attempted approx. 5 installations.  Installing the Arch Linux presents no issues.  I've attached the following: lsblk, grub.cfg as grub.txt and first-boot.log.  I've also included a pict of the boot screen at the failure state.  It appears that the UUID in the failure screen is different from that of the grub file.   This unit is a 2020 MacBook Air M1/8GB/256GB system.  The attempts to install were all made yesterday July 17 and today July 18, 2023.  Thank you.  

[lsblk.txt](https://github.com/UbuntuAsahi/ubuntu-asahi/files/12090032/lsblk.txt)
[grub.txt](https://github.com/UbuntuAsahi/ubuntu-asahi/files/12090036/grub.txt)
[first-boot.log](https://github.com/UbuntuAsahi/ubuntu-asahi/files/12090037/first-boot.log)
[image]

eslerm, 2023-07-19:
closing for https://github.com/UbuntuAsahi/ubuntu-asahi/issues/49

closed 2023-07-19 by eslerm