UbuntuAsahi/ubuntu-asahi #42: Backlight control

itsoulos, 2023-07-05:
Hello,
I have successfully installed Ubuntu 23.04 on my Macbook pro. One minor issue is that I can not control the backlight keyboard. I have installed the light program

Giannis

eslerm, 2023-07-05:
hi Giannis o/

Backlight is not supported in Ubuntu Asahi yet, but will be soon.

[Ubuntu Asahi uses kernel 6.3](https://launchpad.net/~tobhe/+archive/ubuntu/asahi-kernel-build). Upstream landed [backlight control in Linux 6.4](https://lore.kernel.org/lkml/d0862463-acb3-ce83-6493-281a833ec87e@marcan.st/T/#u) and are testing backlight in [asahi-edge](https://github.com/AsahiLinux/docs/wiki/Feature-Support). Once upstream moves this feature into their stable branch, we'll pull in this feature with the new kernel.

closed 2023-07-05 by eslerm

eslerm, 2023-07-14:
Keyboard backlight is working with the labeled keys in Gnome currently. You can also adjust `/sys/class/leds/kbd_backlight/brightness`.