UbuntuAsahi/ubuntu-asahi #43: Some thoughts and some glitches

itsoulos, 2023-07-06:
Dear all,
I have successfully installed Ubuntu 23.04 on my Macbook pro m1 (2020) and everything works perfect for now. Some thoughts
1) I had to pass the parameter --no-sandbox for the Visual Studio code and Microsoft teams to work
2) I had to create a swap file to increase the speed of the machine
3) I have installed my GoogleDrive files using the overgrive package (Insync does not work for me )
4) There are some glitches and traces when I move the mouse in Kde (kwin - wayland). Is there something that I should do in order to avoid this?

Regards
Giannis

tobhe, 2023-07-06:
Hi @itsoulos, glad it works for you!

Some answers and reasons why things don't yet work out of the box:
1. This is probably a issue with the graphics driver not being picked up by the program. are those native or snaps?
2. Good point, we should probably enable that by default
3. Not sure why insync doesn't work
4. I believe that this will be fixed in the next mesa update

While you are already here, there is something you could help us with. We have recently received reports about issues with the 23.04 installer. Could you share the output of `sudo journalctl -u first-boot` here to help us better understand why it works on some machines but not on others?

itsoulos, 2023-07-07:
[first_boot.txt](https://github.com/UbuntuAsahi/ubuntu-asahi/files/11974989/first_boot.txt)
Please find attached the boot log. Also, here is a list of snaps in my machine
snap list
Name                       Version           Rev    Tracking         Publisher      Notes
bare                       1.0               5      latest/stable    canonical✓     base
core22                     20230629          807    latest/stable    canonical✓     base
cups                       2.4.6-2           975    latest/stable    openprinting✓  -
gnome-42-2204              0+git.ff35a85     122    latest/stable/…  canonical✓     -
gtk-common-themes          0.1-81-g442e511   1535   latest/stable/…  canonical✓     -
snap-store                 41.3-71-g709398e  963    latest/stable/…  canonical✓     -
snapd                      2.59.5            19459  latest/stable    canonical✓     snapd
snapd-desktop-integration  0.9               85     latest/stable/…  canonical✓     -

tobhe, 2023-07-07:
Thank you, that helps a lot!

closed 2024-02-18 by eslerm