UbuntuAsahi/ubuntu-asahi #45: jemalloc

dmatora, 2023-07-12:
Does this include asahi jemalloc patch?
Telegram crashes when I try to launch it.

https://github.com/telegramdesktop/tdesktop/issues/24564
https://github.com/telegramdesktop/tdesktop/issues/24836

tobhe, 2023-07-12:
Thx for the report, we currently don't ship a patched jemalloc but it looks like we should. Ideally we can even upstream the patch into the proper Ubuntu package.

tobhe, 2023-07-13:
So looking at the Ubuntu package build files it looks like they should be 64K aligned on 22.04 and 23.10. Are you using a deb package or a snap? Can you show the output of `readelf -l /path/to/libjemalloc` on your system?

dmatora, 2023-07-13:
I've searched for jemalloc via `find / -name "*jemalloc*" 2>/dev/null`
and it only revealed entries inside `/home/ubuntu/tdesktop/` which holds telegram sources I was trying to build, so It looks like it's not system wide issue
I found [this](https://github.com/telegramdesktop/tdesktop/issues/24564) discussion and it seems like telegram is not planning to support asahi systems 

However when I install telegram on asahi using Discover UI - it doesn't crash, [here](https://github.com/telegramdesktop/tdesktop/issues/26103) are some explanations for that

I wonder if there is a a way to take telegram from original asahi and get it running asahi ubuntu or make instruction for patching build files (here are some [suggestions](https://github.com/telegramdesktop/tdesktop/issues/26485))

The main reason I tried switching from asahi to ubuntu is to be able to install deb files which are more popular for software distribution, particularly any desk is available for arm platforms (Raspberry Pi) as a deb package, but it looks like my assumption was wrong and software selection on ubuntu asahi is not wider, but the opposite. (Failed to install any desk from .deb package as well)

clementperon, 2023-10-08:
Installing with snap show that most libraries are statically linked:

```
ldd /snap/bin/telegram-desktop 
	linux-vdso.so.1 (0x0000ffff23f14000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff23d00000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff23edc000)

```

You could try to build from the source, or fork the repo and bump the jemalloc.

clementperon, 2023-10-08:
Easier is rebuild with snapcraft and Jemalloc disabled

[image]

clementperon, 2024-02-07:
@dmatora it has been published here https://snapcraft.io/telegram-asahi

closed 2024-02-18 by eslerm