UbuntuAsahi/ubuntu-asahi #46: 4K kernel

space192, 2023-07-12:
Is there any way we could get 4K kernel for asahi ubuntu ?

tobhe, 2023-07-12:
Hi @space192! Currently we don't support a 4K kernel and as far as I am aware neither does regular Asahi. Once there is an official release we will certainly consider shipping one for people who plan to use emulators and other applications that won't work with 16k pages.

clementperon, 2023-08-28:
@space192 there is no plan to release a 4K kernel on the Asahi roadmap.

However some folks are trying to chrooting one see: https://fosstodon.org/@slp/110940038936968326

Just by curiosity why do need a 4k kernel

space192, 2023-08-31:
I really want to run some x86 programs with FEX but it requires a 4K kernel to run ahah, and the Asahi team seems to be aware that part of the community wants to run x86 apps as Lina the GPU driver developper already tried a 4K kernel with FEX and was able to run some steam game on

tobhe, 2023-08-31:
I am closely watching the progress on that. It looks like fedora is experimenting with 4k kernels for their upcoming release. Like I said above, I will wait for the official release to make the first step and follow once they decide it is stable enough to ship. Until then building custom kernels shouldn't be too hard for those willing to do their own experiments.

space192, 2023-09-07:
Yeah i will probably not be playing around recompiling my own kernel, As I want ubuntu-asahi to be my daily driver x)

clementperon, 2023-10-09:
@space192 this could be interesting. https://sinrega.org/2023-10-06-using-microvms-for-gaming-on-fedora-asahi/

space192, 2023-10-09:
Oh thanks for the article, it looks promising ! seems to be linked with fedora for now, I'll probably try to make it run later this week

tobhe, 2023-10-09:
> @space192 this could be interesting. https://sinrega.org/2023-10-06-using-microvms-for-gaming-on-fedora-asahi/

I haven't had time to try this yet but I would be very interested in learning if there's anything missing in Ubuntu to get it working

closed 2024-02-27 by eslerm