UbuntuAsahi/ubuntu-asahi #47: VSCode / Electron support

clementperon, 2023-07-17:
Hi,

I'm trying to launch vscode 1.80, I also tried with vscode 1.81.

But it seems something is misconfigured by default when it run in sandbox mode it doesn't work :(

```
code-insiders --verbose
[30920:0717/132429.582256:FATAL:credentials.cc(127)] Check failed: . : Permission denied (13)
```

```
code-insiders --no-sandbox
```
Works fine but I got an issue when trying to enable the sync features => it redirects to a webpage but never complete the sync feature

When I turn on the logs it gives me this error
```
rm: cannot remove '/home/ubuntu/snap/chromium/common/chromium/*/GPUCache': No such file or directory
Opening in existing browser session.
libGL error: MESA-LOADER: failed to open apple: /snap/chromium/2543/gnome-platform/usr/lib/aarch64-linux-gnu/dri/apple_dri.so: cannot open shared object file: No such file or directory (search paths /snap/chromium/2543/gnome-platform/usr/lib/aarch64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: apple
```

Clement

tobhe, 2023-08-10:
How did you install vscode? I think this might get fixed soon once we have our own gnome-sdk enabled.

tobhe, 2023-08-11:
Since this is electron I wonder if `MESA_LOADER_DRIVER_OVERRIDE=asahi` and `--ignore-gpu-blocklist` make a difference

tobhe, 2023-08-11:
[image]
`MESA_LOADER_DRIVER_OVERRIDE=asahi /usr/share/code/bin/code --no-sandbox --enable-features=UseOzonePlatform --ozone-platform=wayland`

clementperon, 2023-08-27:
@tobhe sorry for the late reply, for "--no-sandbox" flag.

it seems related to this bug: https://bugs.launchpad.net/ubuntu/+source/linux-meta-nvidia-5.19/+bug/2017980

because the asahi kernel have this feature enabled:
```
sudo cat /proc/sys/kernel/apparmor_restrict_unprivileged_userns
1
```

Maybe it should be a nice thing to disable it to track the same behavior as Lunar kernel no ?

tobhe, 2023-08-27:
Interesting! I would recommend keeping that option enabled since it really helps protect against many potential local root exploits but this means we can also just add an apparmor config which explicitly allows code to use them. Let me see if I can get it running.

tobhe, 2023-08-27:
Oh you are right, it isn't supposed to be enabled in lunar. This is because we are tracking a newer kernel and rebasing the changes from mantic was easier. I agree that we should probably disable it then, there is probably a lot more that won't work without manual changes in lunar.

clementperon, 2023-08-28:
@tobhe ok it makes sense, could it be possible to try to rebase the asahi kernel on the "new" lunar 6.2?

clementperon, 2023-10-08:
Fixed with 6.5.0-1005-apple-arm thanks a lot @tobhe <3

closed 2023-10-08 by clementperon

hamza72x, 2023-11-05:
```sh
MESA_LOADER_DRIVER_OVERRIDE=asahi /opt/brave.com/brave-beta/brave-browser-beta --no-sandbox --enable-features=UseOzonePlatform --ozone-platform=wayland

[24505:24505:1105/074745.108718:ERROR:policy_logger.cc(157)] :components/enterprise/browser/controller/chrome_browser_cloud_management_controller.cc(163) Cloud management controller initialization aborted as CBCM is not enabled.
[1105/074745.135800:ERROR:elf_dynamic_array_reader.h(64)] tag not found
/opt/brave.com/brave-beta/brave-browser-beta: line 48: 24505 Segmentation fault      (core dumped) "$HERE/brave" "$@"
```

@tobhe