UbuntuAsahi/ubuntu-asahi #50: Installation on M1 mac mini stuck at extracting disk.img

rohityadavcloud, 2023-07-23:
I've a M1 mac mini with latest software installed. I'm not able to get the installer working fast, it seems to be stuck and going really slowly when trying to extract disk.img to disk07. The installer.log don't suggest anything, and it isn't trying to download anything. Any pointers on how to debug further?

tobhe, 2023-07-23:
Hi @rohityadavcloud, this sounds like it might be a problem with your connection to the file server. The images are hosted at https://files.tobhe.de/ubuntu/os/. Do you have the same problem when using curl to fetch them? Long shot, but another thing you could try is disabling IPv6 before running the installer.

rohityadavcloud, 2023-07-23:
Yup, that was it. I created a repo mirror in a local datacenter and it worked fast. Thanks for replying @tobhe 

I'm testing to support M1 Mac Mini as KVM guest to use with CloudStack (I did an old experiment with arm64 RPi4 https://rohityadav.cloud/blog/cloudstack-rpi4-kvm) with M1 Mac Mini as arm64 KVM host. Few suggestions for the Ubuntu 22.04 server image to (a) setup netplan in dhcp by default for the default nic, (b) install openssh-server with password auth enabled.

Closing on the remarks, thanks again.

closed 2023-07-23 by rohityadavcloud

tobhe, 2023-07-23:
Thanks for the suggestions! I am currently working on moving the 22.04 image creation to the scripts used to create the official Ubuntu images instead of a simple debootstrap + custom scripts which at the moment. Once that is done, I will see how I can improve our default configuration.

runofthemillgeek, 2025-01-19:
Sorry for piling on here but @rohityadavcloud @tobhe — how does one go about creating a repo mirror to speed up downloads? I've left my installation for hours and it only moves at a snail's pace (~50 Kbps) in extracting `.img` steps. I'm guessing the problem I'm facing is similar to the issue here.