UbuntuAsahi/ubuntu-asahi #52: New GPU drivers

cowpod, 2023-07-25:
Hi, does your version of linux-asahi-edge include the newer OpenGL 3.1 drivers? I've installed the latest version, and updated all my packages.

tobhe, 2023-07-25:
Yes, see glxinfo:
```
user@ubuntu:~$ glxinfo | grep OpenGL
OpenGL vendor string: Mesa
OpenGL renderer string: Apple M2 (G14G B0)
OpenGL core profile version string: 3.1 Mesa 23.2.0-devel
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 23.2.0-devel
OpenGL shading language version string: 1.40
```

santurini, 2023-07-26:
@tobhe Are you using a Mac M2 without troubles? I own a MacBook Air with M2 chip and wanted to give it a try, do you suggest it or can tell me something more about it (know issues, everything fine, etc)?

tobhe, 2023-07-26:
The M2 air works just fine, of course some features don't work yet as documented here https://github.com/AsahiLinux/docs/wiki/Feature-Support
For the m2 the most notable things that don't work at the moment are Webcam support, external Display, Speakers and the built-in mic. Other than that everything I use works quite well.
I've been using mine for around a year now and I am pretty happy. There used to be bugs in the ecosystem too because of software that made wrong assumptions about page sizes etc but that has gotten a lot better with more people using Apple Silicon.

santurini, 2023-07-26:
I did the installation (I'm on Asahi now) but I'm still not able to change the user, it keeps appearing the default ubuntu user even if during the installation process it asked for the new user credentials. Should I open a new issue?

tobhe, 2023-07-26:
I don't think our installer allows selecting a user. You can change the user name in gnome settings, it becomes effective after a reboot.

santurini, 2023-07-26:
> I don't think our installer allows selecting a user. You can change the user name in gnome settings, it becomes effective after a reboot.

It actually becomes effective only at the startup, while the terminal still says "ubuntu@ubuntu" but is not a big deal.
Everything works surprisingly well, excellent job!

closed 2023-07-26 by cowpod