UbuntuAsahi/ubuntu-asahi #53: Night Light/Night Shift

cowpod, 2023-07-26:
Supposedly this is fixed with new packages linux-asahi-edge mesa-asahi-edge (and if using KDE plasma-asahi-edge).
Obviously we're not using KDE, and mesa-asahi-edge doesn't exist for us. What can be done to get it working?

tobhe, 2023-08-11:
I think this is gnome using a different GPU feature than KDE and that doesn't work yet. I remeber this beeing discussed on irc before: https://oftc.irclog.whitequark.org/asahi-dev/2022-04-20.
I think gnome is using gamma tables, kde uses a different mechanism.

eslerm, 2024-02-18:
Upstream has this working in KDE Plasma https://github.com/AsahiLinux/docs/issues/67 However Kubuntu does not work, and should have a recent enough kwin.

GNOME issue is open https://gitlab.gnome.org/GNOME/mutter/-/issues/2318

wlroots also has an open issue https://gitlab.freedesktop.org/wlroots/wlroots/-/issues/1078 and will affect gammastep (https://gitlab.com/chinstrap/gammastep/-/issues/54) and wlsunset.