UbuntuAsahi/ubuntu-asahi #55: Cannot install on m2 Macbook Air

AISLVE, 2023-09-02:
dharya@Dharyas-Laptop ~ % curl -sL https://tobhe.de/ubuntu/install > install.sh
dharya@Dharyas-Laptop ~ % less install.sh       
dharya@Dharyas-Laptop ~ % sh install.sh
Bootstrapping installer:
  Checking version...
  Version: v0.6.4
  Downloading...
  Extracting...
  Initializing...

The installer needs to run as root.
Please enter your sudo password if prompted.
Password:

Welcome to the Asahi Linux installer!

This installer is in an alpha state, and may not work for everyone.
It is intended for developers and early adopters who are comfortable
debugging issues or providing detailed bug reports.

Please make sure you are familiar with our documentation at:
  https://alx.sh/w

Press enter to continue.

Collecting system information...
  Product name: MacBook Air (15-inch, M2, 2023)
  SoC: Apple M2
  Device class: j415ap
  Product type: Mac14,15
  Board ID: 0x2e
  Chip ID: 0x8112
  System firmware: iBoot-10151.1.1
  Boot UUID: 69399744-DF9F-46B9-99FA-1D037A4A85E7
  Boot VGID: 69399744-DF9F-46B9-99FA-1D037A4A85E7
  Default boot VGID: 69399744-DF9F-46B9-99FA-1D037A4A85E7
  Boot mode: macOS
  OS version: 14.0 (23A5328b)
  SFR version: 23.1.328.5.2,0
  System rOS version: 14.0 (23A5328b)
  No Fallback rOS
  Login user: dharya

This device is not supported yet!
Please check out the Asahi Linux Blog for updates on device support:

   https://asahilinux.org/blog/

tobhe, 2023-09-02:
Hi @AISLVE, support for the new 15" Macbook air is still work-in-progress in Asahi and thus also in Ubuntu Asahi. I can't tell when it will be ready but we can keep this issue open until it is.

AISLVE, 2023-09-02:
Ok, thanks for your quick response, looking forward @tobhe

eslerm, 2023-10-29:
This is now supported.

closed 2023-10-29 by eslerm