UbuntuAsahi/ubuntu-asahi #58: Updating to latest version breaks booting

AntoniBertel, 2023-09-17:
After updating packages and rebooting, Asahi fails to load due to a Grub exception: "error: no such device" on Linux. The issue arises because the Grub environment variables point to the wrong partition, i.e., (hd0, gpt3) instead of (hd0, gpt4). Although this is fixable, the next step also fails in BusyBox. The environment variable ROOT points to a UUID that doesn't correspond to any existing device.

M1 Pro

tobhe, 2023-09-17:
Ouch, that doesn't sound good. Is this a fresh installation (there have been issues after the first reboot previously)? Which Ubuntu release did you install?

AntoniBertel, 2023-09-18:
I took the simplest approach (curl -sL https://tobhe.de/ubuntu/install | bash), booted into Ubuntu, and then installed all the recommended updates. I reinstalled the OS, and it works just fine now.

closed 2023-09-18 by AntoniBertel