UbuntuAsahi/ubuntu-asahi #59: Requesting 13.5 Firmware for Installer

1Guardian, 2023-09-27:
Hello, I just installed Asahi Fedora on a 2023 MacBook Pro using expert mode. In the fedora installer, firmware 13.5 is supported and can be used to install, but when I tried the Ubuntu Asahi installer it failed on the firmware being too old. I was just wondering if there was any chance we would get an updated installer so that we can install the Ubuntu Asahi remix easier for 13.5 firmware devices.

tobhe, 2023-09-27:
Thanks for the report, I will see what we can do! I think we might need to build a new installer image for that but I am not quite sure. In any case I'll need to do some testing.

1Guardian, 2023-09-27:
If you need me to test at all, just let me know as I have a macbook pro with firmware 13.5 at my disposal!

tobhe, 2023-09-28:
you could test `curl -sL https://tobhe.de/ubuntu/install-13.5 | sh` which has 13.5 enabled in the config

1Guardian, 2023-09-28:
Just tried it, it fails after requesting the root password due to a file not found error for the file "installer_data.json"

tobhe, 2023-09-28:
Indeed, should be fixed now.

1Guardian, 2023-09-28:
Ok, the installer now works, but the keyboard and trackpad do not. The mesa driver is also not loaded, but it also didnt load on asahi fedora, so not sure thats an asahi ubuntu issue. The keyboard and trac did work on fedora though.

tobhe, 2023-09-28:
hm interesting. that is certainly a firmware issue.  can you check dmesg and maybe `ls /usr/lib/firmware/vendor`? I am also not entirely sure if that laptop needs a newer kernel than what is installed by default there.

1Guardian, 2023-09-28:
I can check in a few hours and let you know. I think it does need a newer kernel as the fedora one shipped with 6.5, but I will confirm when I have a compatible internet adapter and can update the kernel on it.

tobhe, 2023-09-28:
I have a 6.5 kernel in https://launchpad.net/~tobhe/+archive/ubuntu/asahi-testing if you'd like to try that. I am planning to release it soon and more testers won't hurt :)

1Guardian, 2023-09-28:
Sure thing, I'll give it a shot tonight when I get a chance to install the wifi driver for an adapter. 

What do you mean by enable testing?

tobhe, 2023-09-28:
enable testing == add the https://launchpad.net/~tobhe/+archive/ubuntu/asahi-testing ppa

1Guardian, 2023-09-28:
Gotcha, I will do this when I get access to a hub to get the internet working and will let you know how it works!

1Guardian, 2023-09-29:
I can't install kernel 6.5 for testing as ubuntu 23.04 is trapped in dependency hell. Kernel 6.5 now requires libc6 2.38, but ubuntu 23.04 only supports 2.37

1Guardian, 2023-09-29:
I did get 6.3.0-13 built though, and that fixed the tracpad and keyboard; it's just the graphics that have no driver, but neither does the fedora build on kernel 6.5 so.....

tobhe, 2023-09-29:
that is probably because the graphics driver in testing is for 6.5. You could try `do-release-upgrade -d` to update to mantic and see if that works or wait a bit until we have figured out 6.5 for lunar.

1Guardian, 2023-09-29:
Upgraded to 23.10, and installed the 6.5 kernel, but still no graphics acceleration

BokBoy, 2023-10-11:
Hey, did you guys figure out the mouse issue??
I'm running MacOS 14.1 Beta and got the same issue. It works but it doesn't let me use the mouse or keyboard.
The only button that works is the power/touch id button where if you click it, it starts the shutdown process.
I tried installing the official boot loader to run with the Ubuntu but it doesn't load.
Can somebody help me please??

tobhe, 2023-10-12:
A new 23.04 image is now available. This one should work with 13.5 as it ships the newest kernel. I would be interested to learn if that fixes your issue.

1Guardian, 2023-10-13:
Updated side note: m2 pro graphics are now working with asahi fedora and the latest kernel, and looking at the merge notes in the asahi kernel, it looks like the next major kernel will support the 2023 macbook pro line completely. 

Additionally, the m2 pro macbooks were removed from "expert mode only" in the asahi installer today.

BokBoy, 2023-10-13:
Yeah, I am using fedora currently since it’s the best so far for kvm and gnome

tobhe, 2023-10-15:
Closing because this should work now. Thank you everyone for the feedback!

closed 2023-10-15 by tobhe