UbuntuAsahi/ubuntu-asahi #62: Custom Kernel Failed to Boot Launch Issue after installing

haayfe, 2023-10-01:
I attempted to install Ubuntu today, I followed this tutorial to the letter https://youtu.be/aXh_d1OixRI?si=hbnr9KKJmEVImJOs APART FROM the second after restarting the operating system where you have to input your username and password. I stupidly completely blanked and realised I couldn't remember my username since I was so used to just putting in a password or fingerprint to login. I clicked "n" to show I didn't want to continue and restarted to check my password. I continued the installation fine and booted into Ubuntu, had a look around and before long realised that unfortunately many of the apps that I hoped I'd be able to use were not supported.
I then looked at the end of the video tutorial which told me how to uninstall safely.
I restarted my Mac and when it booted up the Apple logo flashed a few times before showing "Custom Kernel Failed to Boot". I am still able to get into Mac but I need to hold down the power button and select the volume each time.
I'd really appreciate some help as I'm panicking here. Thanks.

tobhe, 2023-10-01:
you can select the default boot entry in the boot menu (the one you get when holding the power button) by holding alt before selecting any of the entries.

The uninstall process is pretty much just removing the Ubuntu partitions from disk, but be aware that deleting the wrong partitions can lead to data loss and a broken macos system so read the instructions carefully. It is all documented here: https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet

tobhe, 2024-04-29:
Closing this one for now. Let us know if you need any more help!

closed 2024-04-29 by tobhe