UbuntuAsahi/ubuntu-asahi #64: Laggy interface - graphics driver problem?

alex-a404, 2023-10-06:
Hello!
I have recently installed Asashi ubuntu and after installing the new OpenGL drivers it became very laggy- mouse movement and window resizing is by far not as smooth as they used to be and also videos and animations play with a lot of latency. Is this a known issue, and does this have anything to do with the installed graphics drivers?
Thanks a lot!

tobhe, 2023-10-06:
The user land graphics driver must match the kernel version in order to work. At the moment the ones in the archive should be in sync. Try checking for updates and rebooting and see if that helps.

If you can paste the output of `apt list --installed | grep mesa` and `apt list --installed | grep linux` we can also see if everything is up to date.

alex-a404, 2023-10-06:
Hi! Here is the output:
```
ubuntu@localhost:~/Desktop$ apt list --installed | grep mesa

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libegl-mesa0/now 23.2.0~pre20230606-1asahi1.23.04 arm64 [installed,upgradable to: 23.3.0~pre20230904-1asahi1.23.04]
libgl1-mesa-dri/now 23.2.0~pre20230606-1asahi1.23.04 arm64 [installed,upgradable to: 23.3.0~pre20230904-1asahi1.23.04]
libglapi-mesa/now 23.2.0~pre20230606-1asahi1.23.04 arm64 [installed,upgradable to: 23.3.0~pre20230904-1asahi1.23.04]
libglu1-mesa/lunar,now 9.0.2-1.1 arm64 [installed,automatic]
libglx-mesa0/now 23.2.0~pre20230606-1asahi1.23.04 arm64 [installed,upgradable to: 23.3.0~pre20230904-1asahi1.23.04]
mesa-utils-bin/lunar,now 8.5.0-1 arm64 [installed,automatic]
mesa-utils/lunar,now 8.5.0-1 arm64 [installed]
mesa-vulkan-drivers/now 23.2.0~pre20230606-1asahi1.23.04 arm64 [installed,upgradable to: 23.3.0~pre20230904-1asahi1.23.04]
```

```
ubuntu@localhost:~/Desktop$ apt list --installed | grep linux

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

console-setup-linux/lunar,now 1.217ubuntu3 all [installed,automatic]
libselinux1/lunar,now 3.4-1build4 arm64 [installed,automatic]
linux-asahi-headers-6.3.0-13/lunar,now 6.3.0-13.1 arm64 [installed,automatic]
linux-asahi-headers-6.3.0-7/lunar,now 6.3.0-7.7 arm64 [installed,automatic]
linux-asahi/lunar,now 6.3.0.13 arm64 [installed,automatic]
linux-base/lunar,now 4.5ubuntu9 all [installed,automatic]
linux-firmware/lunar-updates,now 20230323.gitbcdcfbcf-0ubuntu1.6 all [installed,automatic]
linux-headers-6.3.0-13-asahi/lunar,now 6.3.0-13.1 arm64 [installed,automatic]
linux-headers-6.3.0-7-asahi/lunar,now 6.3.0-7.7 arm64 [installed,automatic]
linux-headers-asahi/lunar,now 6.3.0.13 arm64 [installed,automatic]
linux-image-6.3.0-13-asahi/lunar,now 6.3.0-13.1 arm64 [installed,automatic]
linux-image-6.3.0-7-asahi/lunar,now 6.3.0-7.7 arm64 [installed,automatic]
linux-image-asahi/lunar,now 6.3.0.13 arm64 [installed,automatic]
linux-modules-6.3.0-13-asahi/lunar,now 6.3.0-13.1 arm64 [installed,automatic]
linux-modules-6.3.0-7-asahi/lunar,now 6.3.0-7.7 arm64 [installed,automatic]
linux-modules-extra-6.3.0-13-asahi/lunar,now 6.3.0-13.1 arm64 [installed,automatic]
linux-modules-extra-6.3.0-7-asahi/lunar,now 6.3.0-7.7 arm64 [installed,automatic]
linux-sound-base/lunar,now 1.0.25+dfsg-0ubuntu7 all [installed,automatic]
pptp-linux/lunar,now 1.10.0-1build3 arm64 [installed,automatic]
util-linux-extra/lunar,now 2.38.1-4ubuntu1 arm64 [installed]
util-linux/lunar,now 2.38.1-4ubuntu1 arm64 [installed]
```

tobhe, 2023-10-06:
> `upgradable to: 23.3.0~pre20230904-1asahi1.23`
That seems to be your issue.

alex-a404, 2023-10-06:
Thanks! I upgraded all the packages and mesa, seems to have fixed the issue.