UbuntuAsahi/ubuntu-asahi #65: No mouse or Keyboard

BokBoy, 2023-10-11:
> Hey, did you guys figure out the mouse issue?? I'm running MacOS 14.1 Beta and got the same issue. It works but it doesn't let me use the mouse or keyboard. The only button that works is the power/touch id button where if you click it, it starts the shutdown process. I tried installing the official boot loader to run with the Ubuntu but it doesn't load. Can somebody help me please??
> 
> Edit: It also does request 13.5 for firmware when the original one asks for 12.6.
> 
> MacOS 14.1 Beta MacBook Air M2, 8gb ram, 8 core CPU and GPU.

I thought I may as want to start a new issue because it isn't related to the other one (#59)

tobhe, 2023-10-11:
Have you tried updating your packages? I will build new images later today, those might work better.

BokBoy, 2023-10-11:
Hey, When I sent it, that was the time that I made the packages. 
Also, if that helps, I'm using 23.04, So it might be a driver issue with the new os...

tobhe, 2023-10-11:
I am working on a new 23.04 image that will include a newer kernel. I'll let you know once it is uploaded.

BokBoy, 2023-10-11:
Tysm!!

BokBoy, 2023-10-11:
@tobhe, would the older version of linux work with mouse??

tobhe, 2023-10-11:
I don't think it will. You need a newer kernel.

tobhe, 2023-10-12:
A new 23.04 image is now available. This one should work with 13.5 as it ships the newest kernel. I would be interested to learn if that fixes your issue.

BokBoy, 2023-10-14:
Hey, thanks for updating the 23.04 image, I'll try it now, just have to run diskutil commands to erase fedora

BokBoy, 2023-10-15:
It works, got another issue that I’ll continue and mention this. Thanks.

closed 2023-10-15 by BokBoy