UbuntuAsahi/ubuntu-asahi #67: Question about GPU accelerated ubuntu docker container

space192, 2023-10-26:
So I've been playing around with docker on the fedora remix of asahi linux and wanted to make some test about docker gpu passthrough, but when running an ubuntu 22.04 based docker image I get that i'm missing:
/usr/lib/dri/apple_dri.so
then when I pasthrough the file it complains about missing libLLVM16-so
I did the exact same and I was missing yet another file
my question is what apt repo / package would I need in the docker to get the GPU to work ?

tobhe, 2023-10-26:
On Ubuntu we have `/usr/lib/aarch64-linux-gnu/dri/apple_dri.so` which is part of `libgl1-mesa-dri`. Does that help?

space192, 2023-10-26:
so is it a package that is already available in the standard ubuntu apt repository or do I need to add asahi specific repo to /etc/apt/sources.list ?

tobhe, 2023-10-26:
Ah right, it is part of the UbuntuAsahi ppa for now which is https://launchpad.net/~tobhe/+archive/ubuntu/asahi

space192, 2023-10-26:
perfect thanks, gonna try it out ! :D

space192, 2023-10-26:
[image]
Amazing it works !

closed 2023-10-26 by space192