UbuntuAsahi/ubuntu-asahi #68: boot hangs after updating

JoshAtticus, 2023-10-28:
i updated my system with sudo apt update and now it no longer boots, hanging at this screen every time
[image]

tobhe, 2023-10-28:
Hi @JoshAtticus. To help you figure out what's going on we need a bit more info. Which Ubuntu release are you running? when did you initially install it? When did you last update and do you remember which packages got updated?

JoshAtticus, 2023-10-29:
> Hi @JoshAtticus. To help you figure out what's going on we need a bit more info. Which Ubuntu release are you running? when did you initially install it? When did you last update and do you remember which packages got updated?

Installed it 40 minutes before it broke, Ubuntu Desktop 23.10, updated some asahi linux and asahi apple packages

tobhe, 2023-10-29:
I tired to replicate what you did and couldn't reproduce the problem with a plain 23.10 install and installing updates. I wonder what went wrong there.

Did you install any additional packages did you otherwise reconfigure your system by adding PPAs maybe or removing packages?

JoshAtticus, 2023-10-31:
> I tired to replicate what you did and couldn't reproduce the problem with a plain 23.10 install and installing updates. I wonder what went wrong there.
> 
> Did you install any additional packages did you otherwise reconfigure your system by adding PPAs maybe or removing packages?

I think the system ran sudo apt autoremove at some point, maybe that could be the issue? I will try reinstalling asahi once I get home this arvo

JoshAtticus, 2023-11-01:
Forgot to do it, remind me in 4 hours and I'll do it

JoshAtticus, 2023-11-01:
[image]

doing the thing now

tobhe, 2023-11-01:
Thanks for helping us debug this! I am looking forward to see if it is reproducible or if it was a random hick-up... Currently I still have no idea what might have happened.

JoshAtticus, 2023-11-02:
Found the update that makes boot break

`ubuntu-asahi/mantic 1.16 arm64 [upgradable from: 1.15]`

tobhe, 2023-11-02:
so it happened a second time? all this update should do is add an explicit grub dependency. did you see any other notable error messages?

JoshAtticus, 2023-11-02:
> so it happened a second time? all this update should do is add an explicit grub dependency. did you see any other notable error messages?

yes, it happened a second time, there aren't any other error messages, it just seems to break gdm

tobhe, 2023-11-02:
ok, I will build a new image with that version built in and see if that fixes it.

JoshAtticus, 2023-11-02:
How do I update to it when it releases?

JoshAtticus, 2023-11-05:
Hello, any updates?

JoshAtticus, 2023-11-07:
It's been 5 days, any updates?

tobhe, 2023-11-07:
Sorry for the late reply @JoshAtticus , I have been busy traveling for the last few days.
I have uploaded a new image so fresh installations _should_ not have the bug.

For your existing installation you could try to repair it from a linux usb disk. https://arvanta.net/alpine/install-alpine-m1/ documents how to create one based on alpine and how to boot it from alpine.

I also haven't yet understood why this happened only to you. Did you do anything unconventional between installing, updating and rebooting? Did you install or remove additional packages?

JoshAtticus, 2023-11-07:
> Sorry for the late reply @JoshAtticus , I have been busy traveling for the last few days. I have uploaded a new image so fresh installations _should_ not have the bug.
>
> For your existing [...]

It's ok, and no, I haven't done anything I don't normally do. I installed Vivaldi browser, ran sudo apt update && sudo apt upgrade and it broke after reboot.

tobhe, 2023-11-07:
I would recommend you wait a bit, I am currently testing a new image because the previous one seemed to cause a different bug. I can ping you here once I know it works on my machine.

JoshAtticus, 2023-11-07:
> I would recommend you wait a bit, I am currently testing a new image because the previous one seemed to cause a different bug. I can ping you here once I know it works on my machine.

Alright thanks

tobhe, 2023-11-07:
Should be fixed now, tested fresh install and `apt update` + `apt upgrade` on a m2 air. Let me know if you still encounter errors!

JoshAtticus, 2023-11-16:
Haven't reflashed but I am no longer having the issue. Will reopen if it happens again!

closed 2023-11-16 by JoshAtticus