UbuntuAsahi/ubuntu-asahi #69: Notch Support

eslerm, 2023-10-29:
Enable `apple_dcp.show_notch=1` on DE's which support it. So far, none do out of box.

See https://www.reddit.com/r/AsahiLinux/comments/17ci7fh/complete_camera_notch_compatibility_on_gnome/ and https://www.reddit.com/r/AsahiLinux/comments/zibh2f/is_the_m2_macbook_air_notch_area_unused/

BokBoy, 2023-10-30:
I guess it's possible but it may loose some data. 
You know how when you move your cursor underneath the notch, it disappears??
The OS (Linux and MacOS) think they are usable pixels and don't make that place blank.
Especially in Ubuntu, the time is placed there so It may disappear.
I'd love to see it happen and don't take this negatively.

eslerm, 2023-10-30:
^ the GNOME demo looks clean. Ubuntu Asahi won't ship mods, but hopefully support will land upstream in mutter or KDE etc.

In sway I configured waybar around it and I'm fairly happy. The top corners have curves too. It would be nice if fullscreen applications did not use the top of the screen, but good enough for my use.

hamza72x, 2023-11-05:
@eslerm can not even manually enable that. Any instruction?

```sh
Sourcing file `/etc/default/grub'
/usr/sbin/grub-mkconfig: 11: /etc/default/grub: apple_dcp.show_notch=1: not found
```

eslerm, 2023-11-05:
There are a few ways, but I edited `/etc/default/grub` and added the flag to `GRUB_CMDLINE_LINUX_DEFAULT`, ran `update-grub` and rebooted.

Like:
```terminal
# cat /etc/default/grub | grep GRUB_CMDLINE_LINUX_DEFAULT
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash apple_dcp.show_notch=1"
```

hamza72x, 2023-11-05:
Worked, Thanks

BokBoy, 2023-11-17:
@eslerm Would it work on any gnome based os??
I started using fedora 39, would it work there?? (its with gnome)

eslerm, 2023-11-18:
The kernel parameter should work regardless of OS.