UbuntuAsahi/ubuntu-asahi #70: Upgrade to 23.10

clementperon, 2023-11-01:
Hi,

Just noticed that 23.10 seems supported but trying to perform an upgrade response that no new release found..

Is there a way to upgrade to 23.10 from 22.04?

```
$> sudo do-release-upgrade
Checking for a new Ubuntu release
No new release found.
```

Thanks for your help

eslerm, 2023-11-01:
Does `do-release-upgrade --devel-release` help?

clementperon, 2023-11-01:
Thanks @eslerm, yes it works.

FYI during the upgrade check I got the following error:

> Checking for installed snaps
> error: no snap found for "firmware-updater"

eslerm, 2023-11-02:
Looks like the `firmware-updater` snap is/was a known upstream issue: https://bugs.launchpad.net/ubuntu/+source/ubuntu-release-upgrader/+bug/2039268

Appreciate your reports :)

closed 2023-11-02 by eslerm