UbuntuAsahi/ubuntu-asahi #71: Widevine Support

clementperon, 2023-11-01:
Hi,

I have try the widevine installer from marcan with some fixes for Bash, but it seems to not work with Chromium :(

https://github.com/clementperon/widevine-installer

Does someone try and had a success?

Thanks

eslerm, 2023-11-01:
An Asahi only snap or [asahi channel track](https://forum.snapcraft.io/t/track-request-for-gnome-42-2204-asahi-track/36295) in existing browser snaps are likely necessary, unless snaps are not used. We'll talk to the snap folks :)

clementperon, 2023-11-01:
Is it not possible to manually copy the shared libs in the "chromium snap environement"

eslerm, 2023-11-01:
Something like that should work. For Ubuntu Asahi we want an upstream solution, and will help with that.

From https://oftc.irclog.whitequark.org/ubuntu-asahi/2023-11-01 snaps based on core24 may be required for glibc dependencies.

clementperon, 2023-11-26:
@eslerm I try to rebuild a proper chromium snap => switch to core24 => rebuilding gnome-sdk-asahi => rebuild gnome-sdk. So really not trivial, I think I will wait at list that gnome-sdk switch to core24 to go for the proper solution.

For the dirty solution do I have to extract the squasfs -> patch the lib and repackage? Or do you see another solution?

eslerm, 2023-12-06:
Manually copying should work. It can be done through snapcraft too. After unsquashing, you can rebuild and edit `squashfs-root/snap/snapcraft.yaml`. Commands can be ran during different stages etc.

If there is a clean distribution system for asahi-widevine, there's a chance upstream browser's would o-k an asahi snap track.

SevralT, 2024-01-20:
First. Thanks for your fix in script. Second, after hours I found problem. It so easy problem. 
[image]
Configs from file just don't add automatically to config of firefox. You can fix this problem with just manual adding that in about:config

theofficialgman, 2024-02-08:
chromium flatpak should work (once you add Widevine in a place it is looking for) as it switched to a runtime with GLIBC 2.36+ (support for RELR, required by ChromeOS ARM64 Widevine) 3 weeks ago https://github.com/flathub/org.chromium.Chromium/pull/346

orisson003, 2024-08-10:
Just encountered this issue when switching from Fedora to Ubuntu on M1. So far widevine still not working on default Firefox on Asahi Ubuntu. I’m wondering if this needs an upstream fix that snap Firefox needs core24 for the glib dependence. Is it possible to contact the snap folks?

callo90, 2025-08-28:
> So I guess so much time later I have figured out ways to solve this problem: use Brave to get this done. But you do need to do some manual settings to enable DRM/Widevine support in Brave and let it know the Widevine files created by Widevine-installer/Asahi Linux.
> 
> 1. First use non-sandboxed Brave installed from its official website. Sandboxing seems interfering with Widevine according to my experiments.
> 2. Normally non-sandboxed Brave should be installed into /opt/brave.com/brave/. You should run: sudo mkdir /opt/brave.com/brave/WidevineCdm. Then you'll see the DRM option in Brave's settings. Toggle it up.
> 3. The Widevine installer should normally generate widevine files in this folder:  /var/lib/widevine/WidevineCdm. Create the symlink pointing to the actual Widevine folder to let the browser know the location: sudo ln -s /var/lib/widevine/WidevineCdm /opt/brave.com/brave/WidevineCdm.
> 4. Make sure you set the permission right otherwise DRM will still not work: sudo chmod -R a+rX /var/lib/widevine/WidevineCdm.
> 5. Try Netflix and Spotify! Now it should be working well.

This worked, Sandboxes 

> So I guess so much time later I have figured out ways to solve this problem: use Brave to get this done. But you do need to do some manual settings to enable DRM/Widevine support in Brave and let it know the Widevine files created by Widevine-installer/Asahi Linux.
> 
> 1. First use non-sandboxed Brave installed from its official website. Sandboxing seems interfering with Widevine according to my experiments.
> 2. Normally non-sandboxed Brave should be installed into /opt/brave.com/brave/. You should run: sudo mkdir /opt/brave.com/brave/WidevineCdm. Then you'll see the DRM option in Brave's settings. Toggle it up.
> 3. The Widevine installer should normally generate widevine files in this folder:  /var/lib/widevine/WidevineCdm. Create the symlink pointing to the actual Widevine folder to let the browser know the location: sudo ln -s /var/lib/widevine/WidevineCdm /opt/brave.com/brave/WidevineCdm.
> 4. Make sure you set the permission right otherwise DRM will still not work: sudo chmod -R a+rX /var/lib/widevine/WidevineCdm.
> 5. Try Netflix and Spotify! Now it should be working well.

This worked. The issue seems to be with sandboxed apps. I tried the same approach with Flatpaks but it didn’t work. Installing the non-sandboxed version of the app seems to be the way to go.

tobhe, 2025-08-28:
I did fork https://github.com/tobhe/widevine-installer for making it work with the snap versions of firefox and chromium at some point but I found another issue in the process: the widevine binary from chromeos is linked against a rather recent version of glibc. Sandboxing is not the biggest issue, it is that snaps are usually running in a containered environment based on an older Ubuntu LTS release. widevine won't work with Ubuntu 22.04 so only the candidate channels of the browser snaps which had already switched to core24 work.