UbuntuAsahi/ubuntu-asahi #72: Brave Browser crashing

hamza72x, 2023-11-05:
Hey!

Just cross-creating the same issue. Not sure if it's Asahi related or Brave itself.

Linked Issue: https://github.com/brave/brave-browser/issues/34118

tobhe, 2023-11-05:
Thanks for the report! I saw people report similar issues in the fedora-asahi matrix channel, so it doesn't seem to be Ubuntu specific. I think reporting it upstream was the right thing to do, we can keep this issue here around for others encountering the same.

hamza72x, 2023-11-11:
Temporary Fix to use old version:

```sh
sudo apt install brave-browser-beta=1.59.109
```

For ignoring auto-update for `brave-browser-beta`, use:

```sh
sudo apt-mark hold brave-browser-beta
```

hamza72x, 2023-11-17:
It was `upstream` issue, they fixed. Closing.

closed 2023-11-17 by hamza72x