UbuntuAsahi/ubuntu-asahi #73: Can be installed on an external USB or USB C device

bonna79, 2023-11-06:
I apologize but I don't know where to ask this question.
During the installation phase, is it possible to select an external SSD, connected via USB C or USB 3.1, and run Lunux Ubuntu from the external SSD? without having to partition the disk with mac osx installed?

Thanks and sorry if I posted a question in the wrong place

JoshAtticus, 2023-11-07:
No, Apple Silicon doesn't support booting from external drives

eslerm, 2023-11-07:
@bonna79 u-boot supports external drives. Please see https://github.com/leifliddy/asahi-fedora-usb

tobhe, 2023-11-07:
@eslerm right, usb booting is possible but it would still require resizing apfs to store a recovery and the initial u-boot setup.

It might not work for @bonna79 as they said they do not want to reformat their internal disk. I think what @JoshAtticus meant to say was that it isn't supported by Apple, so not possible without modifications.

JoshAtticus, 2023-11-07:
> @eslerm right, usb booting is possible but it would still require resizing apfs to store a recovery and the initial u-boot setup.
>
> It might not work for @bonna79 as they said they do not want to [...]

yes, that is correct

bonna79, 2023-11-07:
> @eslerm right, usb booting is possible but it would still require resizing apfs to store a recovery and the initial u-boot setup.
>
> It might not work for @bonna79 as they said they do not want to [...]

Interesting . So now I read the specific link and then try to figure out how to make a Linux disk boot from usb .

Thank you

1Guardian, 2023-11-15:
It is possible to do as it's what I use 24/7 on my macbook. 

I use a fedora install to provide an on-disk grub env, and then I flashed the ubuntu image to my 500gb USB c drive and fixed all the mounting issues.

The only bug I have is that during the boot sequence, you have to unplug the usb and plug it back in or youll get kicked for the rootfs not being able to mount after grub.