UbuntuAsahi/ubuntu-asahi #74: EFI boot manager: Cannot load any image

Raikerian, 2023-11-07:
Hi, installing your latest Ubuntu 23.10 Desktop using this command:
```
curl -sL https://ubuntuasahi.org/install | bash
```

Installation itself is successful, however ubuntu fails to boot during first boot (no restart help) with these errors:

[image]

Is there anything extra I am missing?

Also I would like to eventually build/install [ubuntu 23.10 server](https://cloud-images.ubuntu.com/releases/mantic/release/), but currently your scripts work mostly with your predefined livefs images. Is there any guide on how to build those myself or do they not modify anything from the original ubuntu?

tobhe, 2023-11-07:
Hi @Raikerian, thanks for the report! something might have gone wrong with the latest image build. I will see if I can reproduce the error and let you know once I have more info.

tobhe, 2023-11-07:
@Raikerian I uploaded a new version and ran a few tests. Looks like the issue should be fixed now. Thanks again for the report, this was very helpful!

To remove your broken installation you can follow the instructions in https://github.com/AsahiLinux/docs/wiki/Partitioning-cheatsheet

Raikerian, 2023-11-07:
Thank you! Will try, I already nuked it so will just reinstall later.

Any advice how I can package ubuntu server using your scripts by the way? Does it have to be livecd, or can something like https://cloud-images.ubuntu.com/releases/mantic/release/ be used?

tobhe, 2023-11-07:
it isn't trivial I think and livecd-rootfs is a maze not many understand. I might have a chance to ask someone who works on this later today. Ideally we would have to teach launchpad to build server images, the scripts in this repo would just treat it as any other launchpad build.

Raikerian, 2023-11-08:
Thanks, that would be helpful. My end goal is to also be able to network boot from grub (or maybe u-boot) over http, which seems to be not possible right now as u-boot doesn't support built-in apple network drivers and grub `net_ls_cards` is empty due to that. Using ethernet over usb also doesn't seem to help. If you know anything about that, would love to hear.

But as of this issue, its fixed now, thanks again.

closed 2023-11-08 by Raikerian