UbuntuAsahi/ubuntu-asahi #75: DisplayLink on 22.04

EdvardSire, 2023-11-17:
Hi,
First off thank you for this repo!

I've followed through with the installer and I'm trying to get DisplayLink to work like in the image shown here: [`working demo`](https://www.reddit.com/r/AsahiLinux/comments/12pvfzn/displaylink_and_ubuntu_on_m1/)

From what i understand i need both Mesa and asahi-edge kernel.
I dug around some other issued and from what I can grep out of apt i already have mesa.

How can i switch to the asahi-edge kernel? Or can you point me in the right direction?:)

eslerm, 2023-11-18:
Please let us know if DisplayLink works for you.

(*DisplayPort* support is on upstream's roadmap.)

EdvardSire, 2023-11-18:
Following this [guide](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu) did not work for me. This guide has always worked for me on non macs. Do you guys have any idea how I can debug this?

eslerm, 2023-11-18:
I'm not too familiar.

The keyring deb file adds broken urls to `/etc/apt/sources.list.d/synaptics.list`:
```
deb [signed-by=/usr/share/keyrings/synaptics-repository-keyring.gpg] https://www.synaptics.com/sites/default/files/Ubuntu/ stable main
deb [signed-by=/usr/share/keyrings/synaptics-repository-keyring.gpg] https://www.synaptics.com/sites/default/files/Ubuntu/ stable non-free
```

EdvardSire, 2023-11-19:
Thank you for the response:)

I removed the keyring and uninstalled `displaylink-driver`. After running the runfile in `Latest Official Driver` and rebooting it's still not working.

[This](https://discussion.fedoraproject.org/t/fedora-37-kernel-6-1-issue-with-lenovo-dock/69370/6) fedora discussion made be realize that there is a difference between the docs. 40AN vs 40AC (usb vs thunderbolt).

I indeed have a 40AC (thunderbolt) and from what i see over at the hardware [docs](https://github.com/AsahiLinux/docs/wiki/M1-Series-Feature-Support) it's still WIP.

And one last thing, can someone help me understand what is meant by this [comment](https://www.reddit.com/r/AsahiLinux/comments/12pvfzn/comment/jgpcnu5/) with regards to mesa?

EdvardSire, 2023-11-19:
Can confirm that it works with a DisplayLink **usb** hub:)

eslerm, 2023-11-19:
\o/ congrats @EdvardSire

hamza72x, 2023-11-25:
I have followed this instruction, it started working immediately.

https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

tobhe, 2024-01-14:
Closing because it looks like the issue was resolved

closed 2024-01-14 by tobhe