UbuntuAsahi/ubuntu-asahi #76: Screen Resolution

trfhgx, 2023-11-17:
Operating System:

macOS version: Version 14.1.1
Ubuntu version: Latest
Laptop model: M1 macbook pro 13inch
Issue:
I successfully installed Ubuntu on my MacBook alongside macOS, but I'm experiencing difficulties with the display resolution. my laptop works only on certain screen resolutions, and when I try to boot from Ubuntu, neither the external nor the main display works properly 

Is there a method for me to manually edit a custom resolution for the main display in Ubuntu from my primary macOS environment by modifying specific files within the Ubuntu volume?

please help

eslerm, 2023-11-18:
HDMI out for external displays is not yet supported https://github.com/AsahiLinux/docs/wiki/M1-Series-Feature-Support#m1-devices

tobhe, 2024-01-14:
Closing because this should now work as confirmed by @timhm on irc.

closed 2024-01-14 by tobhe