UbuntuAsahi/ubuntu-asahi #77: Freezing issue after lock/unlock

hamza72x, 2023-11-24:
Screen/App freezes after lock and unlock (twice), only (top) status bar works and a log-out does reset the freeze.

Steps:

1. Lock (Ctrl+L)
2. Unlock
3. Again Lock
4. Unlock
5. Dash/Dock will disappear, only the last opened app will work. No app will open (from status bar buttons)
6. Now logout by (top) status-bar
7. Unlock/Login
8. Now unfreezes

hamza72x, 2023-11-24:
Normal/Working View:

[image]

Freezed View:

[image]

hamza72x, 2023-11-24:
Normal/Working View

[image]

hamza72x, 2023-11-24:
Freezed View

[image]

tobhe, 2023-11-25:
Hi @hamza72x, I haven't seen this kind of bug before so it's hard to tell what might be causing it. A first possible solution I would recommend is upgrading to 23.10 to see if the problem persists there.

hamza72x, 2023-11-25:
@tobhe using `sudo do-release-upgrade` should be good? Anything else need to do? before proceeding?

tobhe, 2023-11-25:
I think that should be enough. After the update has finished, make sure to that the asahi ppa entry in `/etc/apt/sources.list.d/` is pointing to mantic

hamza72x, 2023-11-25:
```list
deb http://ppa.launchpad.net/tobhe/asahi/ubuntu lunar main
deb-src http://ppa.launchpad.net/tobhe/asahi/ubuntu lunar main
```

this will be:

```list
deb http://ppa.launchpad.net/tobhe/asahi/ubuntu mantic main
deb-src http://ppa.launchpad.net/tobhe/asahi/ubuntu mantic main
```

tobhe, 2023-11-25:
yes!

hamza72x, 2023-11-25:
Upgraded, same things happening. Not sure what's the problem here.

hamza72x, 2023-11-25:
Forgot to do `sudo apt update + upgrade` after the dist upgrade.
After doing so, the problem seems fixed. Thanks or your replies. @tobhe

closed 2023-11-25 by hamza72x

hamza72x, 2023-11-25:
Found the actual root cause, it was `Dash to Dock` extension.
During dist upgrade, it was automatically disabled and the system was working as expected.
But when I manually enabled the extension again, the problem starts appearing, so removing this for now is the solution.